pagermon/pagermon #428: Logged In User options

s3m1s0n1c, 2020-07-07:
Some options for users who are logged in.

- Message Limit - Set there own message limit.
- Highlights - Custom Highlights for user.
- Filters - Custom filters for certain brigades that user can set by default.
**Agency** - If they wanna just see a specific agency
**Alias** - If they want to see specific alias.., Eg. Tuggerah RFS & Berkeley Vale RFS
Maybe a type in box that looks up the alias table for the names and they can select multiple alias?

I'm sure there's more to add but thats all i can think of atm

s3m1s0n1c, 2020-07-07:
This could be as easy as running a search query when they are logged in and have the option set..

Eg ?alias=Tuggerah RFS,Berkeley Vale RFS

s3m1s0n1c, 2020-07-07:
I guess the filters idea has been kinda been discussed in https://github.com/pagermon/pagermon/issues/230

eopo, 2020-07-15:
Or #410.
Maybe admins can create filters for the whole system and select if user groups (future)/normal users can see and use them.
Users can create filters, that only themselves can see and use.