pagermon/pagermon #429: Word filter for notifications to Telegram

marijn390, 2020-07-19:
I use PagerMon to receive messages from the Dutch emergency services. All notifications arrive well and are forwarded to a Telegram channel. We only want to receive notifications in Telegram of a specific area. Can an option be built in that notifications with certain words will only be pushed to Telegram? This is very important to us because we now receive useless notifications.

112bzo, 2020-07-20:
I also use it for telegram. but what is your isseu then?
I only see reports of cap codes that I wish for. What then word filters for?

marijn390, 2020-07-20:
As I see you are also Dutch so you probably know 112pers. They have a word filter that you can set. Only reports with certain words will come through or not. For example, we only want reports from Vlissingen. Now I cannot filter that because the ambulances come to multiple places in Zeeland. So I would like to set a word filter so that the messages with certain words only come in telegram.

DanrwAU, 2020-07-20:
Has been discussed previously. 

The best way would be to implement globally for all notification plugins, but not something that anybody has started, Sorry.

DanrwAU, 2020-07-20:
Duplicate of https://github.com/pagermon/pagermon/issues/109

closed 2020-07-20 by DanrwAU