pagermon/pagermon #431: Date format and translation

marijn390, 2020-07-22:
I would like to have the options to translate certain names in PagerMon into a language of my own. Also, the date format is now on MM-DD-YY H: i. I would like to be able to adjust this to the date format of NL (DD-MM-YY H: i).

davidmckenzie, 2020-07-22:
Date format is the international standard `YYYY-MM-DD`, not sure where you'd be seeing mm-dd-yy?

marijn390, 2020-07-22:
Oh yes of course I mean mm-dd-yym sorry ..

DanrwAU, 2021-01-22:
> I would like to have the options to translate certain names in PagerMon into a language of my own

This can be done with a custom theme. 

Take a copy of the default folder inside the themes directory and edit the files as required :)