pagermon/pagermon #436: Image v0.3.10

112bzo, 2020-08-12:
Dear friends,

When do we see the image v0.3.10 ? like to test this one.

s3m1s0n1c, 2020-08-12:
I'll try and do it next week some time..

Haven't been well so that takes priority

s3m1s0n1c, 2020-08-16:
Currently working on this...

s3m1s0n1c, 2020-08-16:
**- Fixed process-default.json**
 change ``cwd`` to reflect correct  directory on Pi
 put up the ``max_memory_restart`` to 1G

**- cleaned up /var/log folder**

**- installed Pagermon 0.3.10**

**- imaged SD card**

**- zipped image file**

**- Tested Image on RPI4**
### rtl
pi@pagermonpi:~ $ rtl_test 
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR-V3, SN: SDR_V3_03

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6 
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers

### Multimon-ng
pi@pagermonpi:~ $ multimon-ng 
multimon-ng 1.1.8
  (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
  (C) 2012-2019 by Elias Oenal
Available demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10
Enabled demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10
unixinput.c: pa_simple_new() failed: Connection refused
pi@pagermonpi:~ $ 

### Pagermon Server
pi@pagermonpi:~ $ pm2 status
┌─────┬──────────────────┬─────────────┬─────────┬─────────┬──────────┬────────┬──────┬───────────┬──────────┬──────────┬──────────┬──────────┐
│ id  │ name             │ namespace   │ version │ mode    │ pid      │ uptime │ ↺    │ status    │ cpu      │ mem      │ user     │ watching │
├─────┼──────────────────┼─────────────┼─────────┼─────────┼──────────┼────────┼──────┼───────────┼──────────┼──────────┼──────────┼──────────┤
│ 1   │ server           │ default     │ N/A     │ cluster │ 1197     │ 4m     │ 0    │ online    │ 0.3%     │ 63.7mb   │ pi       │ disabled │
└─────┴──────────────────┴─────────────┴─────────┴─────────┴──────────┴────────┴──────┴───────────┴──────────┴──────────┴──────────┴──────────┘
Module
┌────┬───────────────────────────────────────┬────────────────────┬───────┬──────────┬──────┬──────────┬──────────┬──────────┐
│ id │ module                                │ version            │ pid   │ status   │ ↺    │ cpu      │ mem      │ user     │
├────┼───────────────────────────────────────┼────────────────────┼───────┼──────────┼──────┼──────────┼──────────┼──────────┤
│ 0  │ pm2-logrotate                         │ 2.7.0              │ 1182  │ online   │ 0    │ 0.3%     │ 42.0mb   │ pi       │
└────┴───────────────────────────────────────┴────────────────────┴───────┴──────────┴──────┴──────────┴──────────┴──────────┘
pi@pagermonpi:~ $ 

pi@pagermonpi:~ $ pm2 logs
[TAILING] Tailing last 15 lines for [all] processes (change the value with --lines option)
/home/pi/.pm2/pm2.log last 15 lines:
PM2        | 2020-08-16T10:28:55: PM2 log: PM2 home             : /home/pi/.pm2
PM2        | 2020-08-16T10:28:55: PM2 log: PM2 PID file         : /home/pi/.pm2/pm2.pid
PM2        | 2020-08-16T10:28:55: PM2 log: RPC socket file      : /home/pi/.pm2/rpc.sock
PM2        | 2020-08-16T10:28:55: PM2 log: BUS socket file      : /home/pi/.pm2/pub.sock
PM2        | 2020-08-16T10:28:55: PM2 log: Application log path : /home/pi/.pm2/logs
PM2        | 2020-08-16T10:28:55: PM2 log: Worker Interval      : 30000
PM2        | 2020-08-16T10:28:55: PM2 log: Process dump file    : /home/pi/.pm2/dump.pm2
PM2        | 2020-08-16T10:28:55: PM2 log: Concurrent actions   : 2
PM2        | 2020-08-16T10:28:55: PM2 log: SIGTERM timeout      : 1600
PM2        | 2020-08-16T10:28:55: PM2 log: ===============================================================================
PM2        | 2020-08-16T10:28:55: PM2 log: App [pm2-logrotate:0] starting in -fork mode-
[... 38 lines cut ...]
Sampling at 1411200 S/s.
Output at 22050 Hz.

2020-08-16 10:42:11: 0158143: 4    4-4-4
2020-08-16 10:42:16: 1388647: 44111-25
2020-08-16 10:42:21: 0159717: TAKO,RICHARD / B2710709 / BCAS;; / <24 New grn Sheet: 7822-1
2020-08-16 10:42:23: 0159511: TAKO,RICHARD / B2710709 / BCAS;; / <24 New grn Sheet: 7822-1
2020-08-16 10:42:25: 1389808: Helicopter: ARR 20:10 PT TO NICU TROLLEY-92
2020-08-16 10:42:28: 1550499: 82    -82-82
2020-08-16 10:42:30: 1461645: Helicopter: ARR 20:10 PT TO NICU TROLLEY-92
2020-08-16 10:42:31: 1462497: 92-92-92-92
2020-08-16 10:42:34: 1465798: Helicopter: ARR 20:10 PT TO NICU TROLLEY-92
2020-08-16 10:42:37: 1278499: 72    -72-72
2020-08-16 10:42:39: 1465919: Helicopter: ARR 20:10 PT TO NICU TROLLEY-92
2020-08-16 10:42:40: 1702499: 89-   -89-89
2020-08-16 10:42:44: 1465861: Helicopter: ARR 20:10 PT TO NICU TROLLEY-92
2020-08-16 10:42:46: 1242778: 58209-1
^CSignal caught, exiting!

User cancel, exiting...
Signal caught, exiting!
Reattached kernel driver
pi@pagermonpi:~/pagermon/client $ 

[image]

s3m1s0n1c, 2020-08-16:
Edited 0.3.10 released pager to include Raspberry Pi Image details
Uploaded 2020-08-17-pagermonpi_v0.3.10.zip to Releases

s3m1s0n1c, 2020-08-16:
@112bzo This has now been completed..

Please see - https://github.com/pagermon/pagermon/releases/tag/v0.3.10

Thanks
Sonic

closed 2020-08-16 by s3m1s0n1c