pagermon/pagermon #437: Invalid Time & date messages

crazyfella1, 2020-09-19:
Hi there.

I have the pager on running on Ubuntu successfully. I have managed to get the messages from PDW in Windows to send the messages using the pdw2pagermon python script. Upon receiving messages I am getting invalid date and invalid time for the messages. Does anyone one have any knowledge on what format pagermon needs to be valid date and time. Both machines are in the correct time zone and reading the same date and time.

Thanks for your time and input.

davidmckenzie, 2020-09-20:
It might be trying to parse a timestamp that's not actually a timestamp - try setting `useTimestamp` to false in the configs.

crazyfella1, 2020-09-21:
Yea i had a look at that, because im running the python script there isn't the option to do that.

marshyonline, 2020-09-21:
Can you provide screenshots of the settings you are using in PDW?

crazyfella1, 2020-09-21:
[image]
[image]

jjeffhendryx, 2020-09-27:
[image]
Change it so the When line reads as follows in the python script

crazyfella1, 2020-09-27:
Thanks for that this solved my issue much appreciated!

closed 2020-09-27 by crazyfella1