pagermon/pagermon #438: Mobile view doesn’t show Alias name

AndrewJ1990, 2020-09-25:
**Is your feature request related to a problem? Please describe.**
When viewing on a mobile device the Alias column is not shown, thus not providing the name of the alias given to the capcode.

**Describe the solution you'd like**
Alias to be shown as a column, or as an option

**Describe alternatives you've considered**
An option to not show it, or shown at the end of the message for example as to not take up a column

**Additional context**

[image]

DanrwAU, 2020-09-25:
Correct.

The columns are dynamic based on the width of the device screen. 

Try rotating your device 😉

AndrewJ1990, 2020-09-25:
Tried that but still doesn't change. Is there anyway else for it to show them?

[image]

DanrwAU, 2020-09-25:
Your device resolution is likely below the threshold to trigger the change.

Try "View desktop site" if your browser has it.
This disables the adaptive columns.

AndrewJ1990, 2020-09-25:
I've tried that as well but makes no difference unfortunately. I'm using an iPhone X. If it can't be done that's fine, just a tad annoying

DanrwAU, 2020-09-25:
I don't have an iPhone X so can only bae my tests off a test browser the supposedly emulates it and it seems to work.

Are you using some form of zoom above standard?

AndrewJ1990, 2020-09-25:
No, all normal views and on iOS 14. It however works fine on my iPad Pro 11"

davidmckenzie, 2020-09-25:
Sounds like something screwy in iOS Safari not communicating the viewport size to the CSS renderer. Would be keen to see other iPhone observations - I'm also android.

s3m1s0n1c, 2020-09-25:
What happens if you use chrome or another browser?

s3m1s0n1c, 2020-09-25:
Alias doesnt show on Iphone 8 with IOS 14

Its below the viewport max-width

AndrewJ1990, 2020-09-25:
I just tried on Firefox. The same, and requesting desktop site made no difference.

DanrwAU, 2021-01-22:
So this one is a quirk in the iPhone X and above - the device reports a smaller screen size to browsers (because of the notch) and makes it not show. 

Based on my testing fixing this and showing all columns is useless as the message is squashed. 

I'm about to open a PR to remove the noSmall checks on the alias column to fix just that one.

PR #450 (2021-01-22): Fix Alias Column on iPhoneX
# Description

Removes the small screen detection from the Alias column to show this when iPhoneX in landscape mode. 

Will still hide the column when overall device width is lower than 768px.

Fixes #438 

## Type of change

- [x] Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?
In production .... duh

Blisk browser to emulate iPhoneX

Portrait
[image]

Landscape
[image]

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particul [... cut]
closed 2021-01-22 by marshyonline