pagermon/pagermon #440: Alias Import Duplication

profcamseo, 2020-10-02:
**Describe the bug**
After exporting to bulk edit aliases quick and reimporting it has created a duplicate of most aliases (for me around 2,500)

**To Reproduce**
Steps to reproduce the behavior:
1. export alias csv
2. edit alias colum
3. import cdv

**Expected behavior**
only alias name to be updated, not new ones created, only edited 50 aliases but duplicated over 2,000

**Environment information:**
 - OS: Rasberry Pi

**Additional context**
Currently can't select then shift to scroll down and select last alias to bulk select either for quick fix in the alias page and have to select individually to delete

DanrwAU, 2021-01-22:
I haven't seen this one before, any chance i can get a copy of your export and import csv's to compare and see why it's not picking them up?

My gut feeling is you edited them in excel and dropped the leading 0 off the from of the capcode...

J1GGLYPUFFS, 2021-01-22:
Replicated it for interest's sakes

DanrwAU is correct-
The export CSV when opened in Excel drops the leading zero/s. This results in the import process thinking it's a new record on re-import due to the primary key being different and inserting a new record without the leading zero/s.

profcamseo, 2021-01-26:
closing this task, thankyou @DanrwAU @J1GGLYPUFFS 

For anyone that finds this for future reference and imported already and created new records just edit the database table and delete everything after a certain line (the last manual addition before trying the import) and this will clean up your alias list

closed 2021-01-26 by profcamseo