pagermon/pagermon #445: Gotify Plugin Not Functioning

stubbers, 2020-11-16:
**Describe the bug**
The Gotify plugin is stripping the second forward slash from the protocol part of the Gotify URL

`[pmon]  2020-11-16 02:35:41  debug : Gotify: Sending to https:/gotify.example.net/message: {"title":"Example Alias Title","message":"Example message","priority":5}`

The issue occurs regardless of whether you include the protocol or not in the configuration and even if you add extra slashes.

I believe the case is the use of path.join:

`url = path.join(url, 'message'); // Append /message to URL`

I can find nothing in the NodeJS docs that state path can be used with URLs (ref - https://nodejs.org/api/path.html).

**To Reproduce**
Steps to reproduce the behavior:
1. Configure the Gotify plugin including activitating it for an alias
2. Monitor logs for the above message that shows the incorrect URL

**Expected behavior**
URL does not have the second forward slash stripped

**Screenshots**
N/A

**Environment information:**
 - OS: Docker Image
 - Other details:

**Additional context**
I've never managed to get the Gotify plugin to work but debugged the reason why recently

stubbers, 2020-11-16:
Feel free to assign this to me, I'm testing a fix and once I'm satisfied I'll put in a pull request

stubbers, 2020-11-16:
After patch same config results in the proper URL and a working Gotify plugin

DanrwAU, 2021-01-22:
Fixed in #446

PR #446 (2020-11-16): Fix Gotify Plugin not Working
# Description

Changed path.join to just appending using strings in the Gotify plugin as path.join doesn't seem to like URLs

Fixes #445 

## Type of change

Bug fix (non-breaking change which fixes an issue)

# How Has This Been Tested?

Testing using own instance of Gotify and Pagermon
closed 2021-01-22 by DanrwAU