pagermon/pagermon #448: add filters

mjs1987, 2020-11-18:
i am trying to import filter with filter.ini

this command

cat samples/filters.ini | node import.js --pdw

output stops by 30 items

`
root@pagermonpi:/home/pi/pagermon/client# cat samples/filters.ini | node import.js --pdw
Sending capcode: {"address":"0126900","alias":"Reddingsbrigade )","agency":"Ambulance Kennemerland ( OvD-RB","color":"green","icon":"question"}
Sending capcode: {"address":"1420434","alias":"OMSG Piket )","agency":"Ambulance Rotterdam Rijnmond ( GHOR Functionaris","color":"green","icon":"question"}
Sending capcode: {"address":"0806360","alias":"TS/HV 634 )","agency":"BRW Aalten ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806389","alias":"TS 9032 )","agency":"BRW Borculo ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806390","alias":"TS/HV 609-639 )","agency":"BRW Borculo ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"1170022","alias":"St. Michielsgestel / St. Oeden","agency":"BRW Brabant Noord ( OvD Kring 2 B","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806666","alias":"WOA 8312 )","agency":"BRW Brummen ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502673","alias":"DA-OD 60-94 )","agency":"BRW De Wolden ( OVD ZW","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502511","alias":"DA 20-93 CAC )","agency":"BRW Drenthe ( Assen-regio","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502521","alias":"OWK 20W11 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502522","alias":"OWK 20W12 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502523","alias":"OWK 20W13 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502524","alias":"OWK 20W14 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502525","alias":"OWK 20W21 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502526","alias":"OWK 20W22 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502527","alias":"OWK 20W23 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502528","alias":"OWK 20W24 )","agency":"BRW Drenthe ( Regio NM","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806394","alias":"TS/HV 642 )","agency":"BRW Eibergen ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502592","alias":"VW 40-89 )","agency":"BRW Emmen ( Regio","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806539","alias":"Veetakel 8871 )","agency":"BRW Gendringen ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806378","alias":"TS/HV 673 )","agency":"BRW Groenlo ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806497","alias":"HV-1 9271 )","agency":"BRW Lichtenvoorde ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806382","alias":"TS/HV 638 )","agency":"BRW Lichtenvoorde ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502671","alias":"DA-OD 60-92 )","agency":"BRW Meppel ( OVD ZW","color":"darkred","icon":"question"}
Sending capcode: {"address":"1400763","alias":"HV 732 )","agency":"BRW Middelharnis ( HV ploeg","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806399","alias":"TS/HV 837 )","agency":"BRW Neede ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"1400849","alias":"IM 90.x - Informat","agency":"BRW Rotterdam Rijnmond ( Functionaris","color":"darkred","icon":"question"}
Sending capcode: {"address":"1400848","alias":"IP 90.1 )","agency":"BRW Rotterdam Rijnmond ( Functionaris","color":"darkred","icon":"question"}
Sending capcode: {"address":"1400860","alias":"ROT )","agency":"BRW Rotterdam Rijnmond ( Functionaris","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806425","alias":"TS 746 )","agency":"BRW Silvolde ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0401387","alias":"HV ploeg 456 )","agency":"BRW Stadskanaal ( HW","color":"darkred","icon":"question"}
Sending capcode: {"address":"0502672","alias":"DA-OD 60-93 )","agency":"BRW Westerveld ( OVD ZW","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806478","alias":"HV-2 9371 )","agency":"BRW Winterswijk ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806373","alias":"HW 656 )","agency":"BRW Winterswijk ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0806370","alias":"TS 636 )","agency":"BRW Winterswijk ( MDT","color":"darkred","icon":"question"}
Sending capcode: {"address":"0104992","alias":"Ambtenaar Openbare )","agency":"BRW Zaanstreek Waterland ( AOV Waterland","color":"darkred","icon":"question"}
End of input
Success! 111
Success! 112
Success! 113
Success! 114
Success! 115
Success! 116
Success! 117
Success! 118
Success! 119
Success! 120
Success! 121
Success! 122
Success! 123
Success! 124
Success! 125
Success! 126
Success! 127
Success! 128
Success! 129
Success! 130
Success! 131
Success! 132
Success! 133
Success! 134
Success! 135
Success! 136
Success! 137
Success! 138
Success! 139
Success! 140
Success! 141
Success! 142
Success! 143
Success! 144
Success! 145
Success! 146
root@pagermonpi:/home/pi/pagermon/client#
`

what can i do

davidmckenzie, 2020-11-19:
The import script is pretty unreliable - you'll wanna split it up and do it in chunks probably

mjs1987, 2020-11-19:
Aah Okey,

I see that te import function is not working for csv