pagermon/pagermon #456: PDW Mode - Admin Disable breaking styling on mobile devices

J1GGLYPUFFS, 2021-01-25:
**Describe the bug**
When "PDW Mode - Admin Disable" has been ticked and applied, styling on a mobile device breaks.

Steps to reproduce the behavior:
1. Go to Settings
2. Enable "PDW Mode - Admin Disable"
3. Go back to messages via mobile device and styling appears broken IE: Resizing etc all incorrect and appears to look like a desktop.

Expected behaviour 
For the styling on a mobile device to be the same with or without "PDW Mode- Admin Disable" ticked.

**Environment information:**
 - OS: Raspbian on Pi4

davidmckenzie, 2021-01-28:
* What version was this on?
* Were you logged in on the mobile device when the styling was broken?
* What mobile device? (or if it was on the raspi, what was the screen resolution?)

J1GGLYPUFFS, 2021-01-28:
> * What version was this on?
0.3.10
> * Were you logged in on the mobile device when the styling was broken?
Yes - Logged into mobile device via admin account
> * What mobile device? (or if it was on the raspi, what was the screen resolution?)
Samsung S10+
Samsung S6 Tablet
Iphone X

marshyonline, 2021-01-28:
Interesting - I just tested this on my S10+ (Server Version 0.3.10) and could not replicate the issue with a page of short ish messages

I did however find that if the messages where long - you could zoom out on the mobile page.
Seems text warping is not implemented in mobile views causing ress to not remain the same.

@J1GGLYPUFFS Are you messages long, or sort when you are seeing this issue?
Can you provide us a screenshot?

J1GGLYPUFFS, 2021-02-02:
Hi Marshy,

All messages for the network I monitor are usually quite large in length - Sometimes hitting the networks 496 character limit etc.

Below are two screenshots - I believe you are correct about it being caused by it not wrapping whilst in that mode.
[[image]
](url)

[image]

marshyonline, 2021-02-02:
Thanks for that, it defiantly looks like broken text wrapping - Ill try to look into this soon.

Dtrieb, 2022-01-03:
Here it looks also weird.

[image]