pagermon/pagermon #457: add a field to capcode creation /edition

f5gkw, 2021-01-26:
Add a field on the capcode creation page to place a comment.

Examples :
 "validated on 01 janv 21" or 
 " also linked with capcode xxx" or 
 " owner is john Doe".

and a checkbox on the main page to see there is a comment on this capcode.

Thanks !

DanrwAU, 2021-05-29:
The comment field is added easily enough, The display on the main page is probably a bigger discussion. something for the future

geelongmicrosoldering, 2021-06-01:
> The comment field is added easily enough, The display on the main page is probably a bigger discussion. something for the future

on the alias template, replace
```html
<td class="expand clickable-td" ng-click="messageDetail(alias.id);" style="color: Maroon;">{{alias.alias || ''}}</td>
```
with
```html
<td class="expand clickable-td" ng-click="messageDetail(alias.id);" style="color: Maroon;" title="{{alias.comment}}">{{alias.alias || ''}}</td>
```
obviously 'comment' column has to exist in the capcodes table. renders this on hover:
[image]

You could add the same way in main page, or as part of popover template, but would have to reference an extra column in messages. that could get messy. if all you wanted was a checkbox you could have a boolean "hascomment" column i guess

f5gkw, 2022-07-25:
Hello,
More than one year later, I see this answer ...
Thanks,
I will test

eopo, 2022-07-26:
With "main page", do you refer to the alias list or the message list?

geelongmicrosoldering, 2022-07-26:
> With "main page", do you refer to the alias list or the message list?

Im going to be honest. I dont even remember writing this and would probably do it differently now 😂