pagermon/pagermon #459: Message search returning nil results via text matching

J1GGLYPUFFS, 2021-02-02:
**Describe the bug**
When adding a text matching value IE: Hb or QD, clicking on the highlight of the message returns an ancient result and not the expected list of messages containing Hb etc. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to Admin panel
2. Add an applicable text filter.
3. Return to messages and click the filter.
4. See error

**Expected behaviour**
For all results to be available - Note: I have only observed this issue on small values (IE: Hb) with @@ALERT working fine.

**Screenshots**

[image]

[image]

[image]

**Environment information:**
 - OS: Raspberry Pi running Raspbian
 - Pagermon Version: 0.3.10

DanrwAU, 2021-02-02:
This one is an issue with the search string only being 2 characters.

J1GGLYPUFFS, 2021-02-02:
Hi Dan,

Thanks - Understood but I believe it should either work for all applicable results from the query IE: Hb or not at all

IE: If the users sending the message puts a symbol to start of their message then the filter works (see below)

[image]

davidmckenzie, 2021-02-02:
It's due to the way searches tokenise strings, iirc it splits a string into a chunk of 3 or 4 characters and strips out special characters to form "words". We might have to artificially limit search string characters to prevent any results from showing. (different DBs probably react differently to this, too)

closed 2021-04-11 by J1GGLYPUFFS