pagermon/pagermon #464: Pm2 not functioning for client

Retartist, 2021-03-06:
**Describe the bug**
Trying to start reader.sh in pm2 does nothing, upon examination of the logs it appears that pm2 is reading from the wrong directory

**To Reproduce**
Steps to reproduce the behavior:

1. in console `pm2 start ./pagermon/client/reader.sh`
2. pm2 says it's started
3. check pm2 logs

```Error: Cannot find module '/home/pi/reader.js'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:981:15)
    at Function.Module._load (internal/modules/cjs/loader.js:863:27)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12)
    at internal/main/run_main_module.js:18:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}
```

**Environment information:**
 - OS: Raspberrypi3b os
 - Other details: -- 

Following is after I changed reader.sh to include the filepath of reader.js, if you don't do that it also tries to find reader.js in the home directory 

```Warning: -o is very buggy
multimon-ng 1.1.8
  (C) 1996/1997 by Tom Sailer HB9JNX/AE4WA
  (C) 2012-2019 by Elias Oenal
Available demodulators: POCSAG512 POCSAG1200 POCSAG2400 FLEX EAS UFSK1200 CLIPFSK FMSFSK AFSK1200 AFSK2400 AFSK2400_2 AFSK2400_3 HAPN4800 FSK9600 DTMF ZVEI1 ZVEI2 ZVEI3 DZVEI PZVEI EEA EIA CCIR MORSE_CW DUMPCSV X10 SCOPE
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Elonics E4000 tuner
Tuner gain set to automatic.
Tuned to 148927100 Hz.
Oversampling input by: 12x.
Oversampling output by: 4x.
Buffer size: 7.74ms
Exact sample rate is: 1058400.010094 Hz
internal/fs/utils.js:230
    throw err;
    ^

Error: ENOENT: no such file or directory, open './config/config.json'
    at Object.openSync (fs.js:457:3)
    at Object.writeFileSync (fs.js:1282:35)
    at Object.<anonymous> (/home/pi/pagermon/client/reader.js:20:8)
    at Module._compile (internal/modules/cjs/loader.js:1157:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1177:10)
    at Module.load (internal/modules/cjs/loader.js:1001:32)
    at Function.Module._load (internal/modules/cjs/loader.js:900:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12)
    at internal/main/run_main_module.js:18:47 {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT',
  path: './config/config.json'
}
Sampling at 1058400 S/s.
Output at 22050 Hz.
Signal caught, exiting!

User cancel, exiting...
Signal caught, exiting!```

s3m1s0n1c, 2021-03-07:
@Retartist is this this pagermonpi image or are you getting the repo using git clone?

Did you do a npm install in the client folder?

Did you change the file structure at all? as it needs to as per the repo..

Reader.sh needs other files to run and they need to be in the correct path

Retartist, 2021-03-07:
This is a git clone of the repo, 
Npm install was done of the client folder, I haven't changed anything from the repo. 
Reader.sh works as is if I launch it in shell, just not launching it with pm2

Starting it from inside the folder does work, it's odd that it didn't work the other way.
Maybe add this troubleshooting to the wiki? 

Thanks for your help, solved

s3m1s0n1c, 2021-03-07:
No worries..

closed 2021-03-07 by s3m1s0n1c