pagermon/pagermon #472: Problems with decoding

ChrisX1809, 2021-05-23:
Hey,
I think not directly a bug, but my pagermon doesnt decode messages and i dont know why. 
I use the RPi Image from the docu and installed everything. Pagermon was started with this advises:

https://github.com/pagermon/pagermon/wiki/PagermonPi-Image-For-Raspberry-Pi#configure-pagermon-client-config

Pagermon shows everthing running:
[image]

I dont know where the problem is, could anyone help?

davidmckenzie, 2021-05-23:
Try working forwards from the source - if you tune to the pager freq in something like sdrsharp with same device and antenna, do you get a clean and clear signal when messages come through? If so, then you'd wanna try experimenting with different settings on the multimon bit of the reader script, if not then you'd need to adjust antenna or find the right frequency etc.

s3m1s0n1c, 2021-05-23:
What does your ``reader.sh`` look like?
Try running ``reader.sh`` without the ``node reader.js``
remove -l 15 from reader.sh

and first try what @davidmckenzie suggested :)

ChrisX1809, 2021-05-23:
> Try working forwards from the source - if you tune to the pager freq in something like sdrsharp with same device and antenna, do you get a clean and clear signal when messages come through? If so, [...]

Tried the device with boswatch without problems, the signal was good. So I think, I should work at the settings for multimon. 

> What does your reader.sh look like?
Try running reader.sh without the node reader.js
remove -l 15 from reader.sh

[image]

ChrisX1809, 2021-05-23:
Tried the thing of your tips, but nothing worked. 
Multimon-ng is included in Pagermon or must install manually on top?

ChrisX1809, 2021-05-24:
Got a step further:
Decoding with manually starting rtl_fm and multimon-ng is running: 
rtl_fm -M fm -f 169.91M -s22050 | multimon-ng -a POCSAG512 -f alpha -t raw /dev/stdin

But no visualtion in pagermon, added the lines above in reader.sh and tried with "node" and without. No difference.

ChrisX1809, 2021-05-24:
At least solved the problem, I used the Rpi Image from Pagermon. But theres no nodejs installed. So I had to install it manually.

closed 2021-05-24 by ChrisX1809