pagermon/pagermon #478: FLEX Decoding has characters at the start of a decoded pager message.

hayden-marsh, 2021-08-03:
**Describe the bug**
Recently upgraded to Ubuntu 20.04, clean installed multimon-ng & pagermon. 

FLEX messages start with a few random characters. "3.0.K|" "3.1.F|" "0.0.C|"

**Screenshots**
https://i.imgur.com/Ay8OQlZ.png

**Environment information:**
 - OS: Ubuntu Server 20.04, latest updates of both PagerMon and Multimon.
 - Other details:

DanrwAU, 2021-10-04:
I dont know of anyone running FLEX, but there are quite a few NZ users running it that haven't reported any issues. 

I would suspect the local network has changed something.

hayden-marsh, 2021-10-04:
This was solved by downgrading the version of multimon by a year or two..
Unsure if something changed with the flex output from multimon that
pagermon receives, but it's been solved now.

DanrwAU, 2021-10-17:
Ahh yeah i do recall something Multimon related causing other issues glad you fixed!

closed 2021-10-17 by DanrwAU