pagermon/pagermon #480: oracledb install issue

HobbyKonijn, 2021-09-04:
Hi,

I following the install guide of @marshyonline but get stucked during installation.

I get following error:
npm ERR! code 87
npm ERR! path /home/pi/pagermon/server/node_modules/oracledb
npm ERR! command failed
npm ERR! command sh -c node package/install.js
npm ERR! oracledb ERR! NJS-067: a pre-built node-oracledb binary was not found for linux arm

https://github.com/pagermon/pagermon/wiki/Tutorial---Production-Ready-Ubuntu,-PM2,-Nginx-Reverse-Proxy,-Let's-Encrypt-SSL,-Pagermon-server

I have already a Flex pagermon for P2000 running but try to set it up at an other raspberry pi.
Does somebody is heaving some tips?

HobbyKonijn, 2021-09-06:
Found same issue during install of latest raspberry pi pagermon img.

ryanunderwood, 2021-09-13:
Try running `npm install https://github.com/oracle/node-oracledb/releases/download/v5.2.0/oracledb-src-5.2.0.tgz
`

HobbyKonijn, 2021-09-13:
@ryanunderwood Thanks, found out also this weekend :)
I added to the package.json following:
"oracledb": "https://github.com/oracle/node-oracledb/releases/download/v5.2.0/oracledb-src-5.2.0.tgz",

And for sqlite3:
"sqlite3": "https://registry.npmjs.org/sqlite3/-/sqlite3-5.0.2.tgz",

Both give me 3 weeks to investigate why it was not working.

Pagermon fire's up now.

DanrwAU, 2021-10-04:
The guide above is written for an x86 ubuntu install, raspbian (arm) is missing some packages (oracledb being the big one)

If you open package.json and remove the oracledb line 

`"oracledb": "^4.2.0",`

It should install correctly when doing npm install. If i remember correctly oracledb package isn't called unless you select oracledb in the dropdown.

closed 2021-10-17 by DanrwAU