pagermon/pagermon #486: Can't login on fresh install

therokh, 2021-11-16:
**Describe the bug**
I cannot login with default credentials on a fresh docker build of 3.11. I am returned to the main page.

**To Reproduce**
- Build the container from the instructions in README
- Try to log in with 'admin' and 'changeme' 

**Expected behavior**
Login should occur

**Environment information:**
 - Running in docker built from the docker-compose and docker files in the repo

**Additional context**
I would suspect this is the same problem as #360 and #158 but the docker-compose file sets USE_COOKIE_HOST to false which should skip the sessSet.cookie.domain assignment?

SloCompTech, 2021-11-18:
After build pagermon v0.3.11 docker image runs without any environment variables just fine. Did you try to comment out environment variables ? Did you try to set HOSTNAME to '' ?

therokh, 2021-11-23:
Yes I tried setting hostname to blank but no difference. The login appears to be successful but the 'Sign In' button remains and I cannot access the admin page.

```
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:55 +0000] "GET /auth/login HTTP/1.0" 200 7199 "https://pagermon.example.com/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:55 +0000] "GET /stylesheets/style.css HTTP/1.0" 304 - "https://pagermon.example.com/auth/login" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:55 +0000] "GET /stylesheets/textAngular.css HTTP/1.0" 304 - "https://pagermon.example.com/auth/login" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:55 +0000] "GET /javascripts/auth.main.js HTTP/1.0" 304 - "https://pagermon.example.com/auth/login" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:55 +0000] "GET /templates/auth/login.html HTTP/1.0" 304 - "https://pagermon.example.com/auth/login" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"del","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":[1637643059628],"__knexQueryUid":"87321a9f-2887-42c6-8132-a2e3f473037b","sql":"delete from `protection` where `lifetime` < ?"}
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"select","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":["brKk87zS4w0CSBl3/hv0/O3s7RL6CKJzbkzd6Nnek+Q="],"__knexQueryUid":"732353d3-de81-4bb9-8b88-da0231a863e5","sql":"select * from `protection` where `key` = ?"}
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"select","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":["brKk87zS4w0CSBl3/hv0/O3s7RL6CKJzbkzd6Nnek+Q="],"__knexQueryUid":"bad68df6-9879-42ea-b527-fc16331cbb1e","sql":"select * from `protection` where `key` = ?"}
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"update","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":[1637658359634,4,1637643059631,"brKk87zS4w0CSBl3/hv0/O3s7RL6CKJzbkzd6Nnek+Q="],"__knexQueryUid":"b626315f-195b-49f0-ad66-72082e2711bc","sql":"update `protection` set `lifetime` = ?, `count` = ?, `lastRequest` = ? where `key` = ?"}
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"first","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":["admin",1],"__knexQueryUid":"7c41466a-f661-4800-84f4-c2fcca1ffd61","sql":"select * from `users` where `username` = ? limit ?"}
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"update","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":["2021-11-23 04:50:59",1],"__knexQueryUid":"caed27fe-d3ea-40e3-adad-b3158a6d5677","sql":"update `users` set `lastlogondate` = ? where `id` = ?"}
pagermon    | [auth]  2021-11-23 04:50:59  debug : Successful login {"id":1,"givenname":"Admin","surname":"","username":"admin","password":"$2a$08$De/aXnQkZIEbQ9p8J22tHuzLltqIbsAxE2CGgRMPLaaIwwHmVrpsu","email":"none@none.com","role":"admin","status":"active","lastlogondate":"2021-11-23 04:50:19"}
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"del","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":["/H/+I9JQ40GZr0tdq2HygPqr3q0zUdhZsdY3eAWQGxs="],"__knexQueryUid":"d8a628c8-8812-48e3-bf03-7093110dd963","sql":"delete from `protection` where `key` = ?"}
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:59 +0000] "POST /auth/login HTTP/1.0" 200 35 "https://pagermon.example.com/auth/login" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:59 +0000] "GET /admin HTTP/1.0" 302 46 "https://pagermon.example.com/auth/login" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:59 +0000] "GET / HTTP/1.0" 304 - "https://pagermon.example.com/auth/login" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:59 +0000] "GET /stylesheets/style.css HTTP/1.0" 304 - "https://pagermon.example.com/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:59 +0000] "GET /stylesheets/textAngular.css HTTP/1.0" 304 - "https://pagermon.example.com/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:59 +0000] "GET /apple-touch-icon.png?v=yyx75bRMkG HTTP/1.0" - - "https://pagermon.example.com/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
pagermon    | [db]  2021-11-23 04:50:59  debug : {"method":"select","options":{},"timeout":false,"cancelOnTimeout":false,"bindings":[1],"__knexQueryUid":"dffedacd-656c-48e7-8461-3a686bb44b8e","sql":"select count(*) as `msgcount` from `messages` where (`alias_id` not in (select `id` from `capcodes` where `ignore` = ?) or `alias_id` is null)"}
pagermon    | init: 1.887ms
pagermon    | (node:223) Warning: Label 'sql' already exists for console.time()
[... cut ...] `ignore`, CASE WHEN NOT capcodes.address = messages.address THEN 1 ELSE 0 END as wildcard from `messages` left join `capcodes` on `capcodes`.`id` = `messages`.`alias_id` where `capcodes`.`ignore` = ? or `capcodes`.`ignore` is null order by `messages`.`timestamp` desc limit ?"}
pagermon    | 10.0.0.10 - - [23/Nov/2021:04:50:59 +0000] "GET /api/messages?limit=&page=1 HTTP/1.0" 304 - "https://pagermon.example.com/" "Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:94.0) Gecko/20100101 Firefox/94.0"
```

therokh, 2022-01-10:
I left this alone for a while and rebuilt the container and it's working now? Not sure if the hostname var change ended up fixing it.

DanrwAU, 2022-01-15:
It was 100% the hostname var, this is only useful if you host multiple sites on the same domain. outside of that it causes issues like this and shouldn't be used.

closed 2022-01-15 by DanrwAU

denperss11, 2023-08-06:
I have the same issue, installed it on ubuntu. 
Fresh install. 
- I can load the page
- When i login the page just loops but no login is done. 

log says that the auth was sucess.