pagermon/pagermon #494: Decode Filter active?

ChrisX1809, 2022-01-13:
Hey,

ive got a problem with decoding messages in Pagermon. Used server in Docker and the client locally on the machine. Client runs with rtl_fm and multimon, but got some greyed messages in the log, that arent visible at the pagermon site. 

Is a standard filter active? Didnt find one.

[image]

I use  PagerMon 0.3.11-beta, multimon-NG latest.

takenalias, 2022-01-13:
I'm pretty sure that is because they are not decoding properly. So the
signal is not strong or clear enough or maybe the front end of the SDR is
overloaded.

Less of a pagermon issue and more of a SDR problem.

DanrwAU, 2022-01-15:
The grey messages are just function codes, which are filtered by default. 

By default the pagermon client will only send alphanumeric messages to the server.

closed 2022-01-15 by DanrwAU