pagermon/pagermon #500: Cloudflare + express causes 520 error.

geelongmicrosoldering, 2022-02-12:
**Describe the bug**
_I actually "fixed" this a while ago. Forgot how, reinstalled, and had to chase it again._

I'm not familiar enough with express/node to even begin to debug this, but commenting out the following fixes it on 3 servers:
`res.render(path.join(__dirname,'themes',theme, 'views', 'global', 'error'), { title: title });`
For someone familiar this probably isn't super difficult to squash.

Basically, while it was never officially supported to my knowledge, I use Cloudflare for my nameserver, CDN, etc. The setup is
`Client > Cloudflare DNS > NGINX Reverse Proxy > Node.js (pagermon)`
The issue comes when users login/logout, or when a session expires. Clients get an error 520 from Cloudflare. Depending where you look, you get a "headers already sent" error.

It basically turns out that the error is caused by the code, intended to render the error. Go figure.

For now, I'm just leaving the render disabled.

**To Reproduce**
Steps to reproduce the behavior:
1. Install Pagermon (any version)
2. Configure NGINX for reverse proxy
3. Configure domain to use Cloudflare DNS
4. Navigate to Domain
5. If you don't immediately get a 520 error, log in/out
6. Bypass the error by navigating to /auth/login directly, or comment out the res.render line in app.js

**Expected behavior**
Unauthorized clients, and clients logging out should be redirected to /auth/login

**Screenshots**
[image]

**Environment information:**
 - OS: Ubuntu 20.04, although the same occurs on Rasbian, 18.04, and on multiple Pagermon releases

geelongmicrosoldering, 2022-02-16:
> This is an interesting one - I don't have this issue behind Apache Proxy + CF. Will need to have a deeper dive into this one.
> 
> Are you running secure mode or just default fresh install?

ah yep, should have noted that.

That's with secure mode enabled

crazyfella1, 2022-05-04:
I also am having a similar issue with Client > Cloudflare DNS > Cloudflare Argo Tunnel > Nodejs (pagermon). The work around heading direct to site /auth/login works fine. I will comment out the line highlighted above see if that makes a difference.

vk2mpj, 2023-02-09:
I too am experiencing this issue now. Applying the workaround seems to have gotten me back online