pagermon/pagermon #504: Define color and icon once in Aliasses

krisken, 2022-02-21:
Is it possible to set the color and icon once (in Aliasses), and put a dropdown in Aliases to use it??

Eg I use "#00ca00" as color, "ambulance" as icon and safe it as "ambulance".  Then, when I add a new alias, I can use the pre-defined "ambulance" and everything is set to the correct icon and color.

eopo, 2022-02-27:
Interesting Idea.
Do you think of it as a kind of template to make creation of new alias easier, or do you think of it as a reference, so existing Aliases would change the color and or icon, if the definition is changed?

geelongmicrosoldering, 2022-04-27:
> Interesting Idea. Do you think of it as a kind of template to make creation of new alias easier, or do you think of it as a reference, so existing Aliases would change the color and or icon, if the definition is changed?

i love this, alias templates would make my life easier. have you started on this at all?

geelongmicrosoldering, 2022-04-27:
@eopo assuming you haven't started it and I didn't just step on your toes (noticed you self-assigned)

How about this?

https://user-images.githubusercontent.com/14150258/165453056-3336dd30-9fb2-4e15-89fd-0606bd9f5d61.mov

You add/edit filters from Settings, like you would Text Matching. They are stored in config file
[image]

I can submit a PR with this.
Potentially, I can close [this PR](https://github.com/pagermon/pagermon/pull/512) and submit a new one with both helpers

DanrwAU, 2022-12-21:
Fixed in #514

closed 2022-12-21 by DanrwAU