pagermon/pagermon #505: Alias based on message

Quintenplaizier, 2022-02-21:
Is it possible to create a alias besed on a part of the message?

So that i can create a filter for ALL capcodes but only whit *PRIO1* in the message.

I hope this is possible.

Kind regards
Quinten

eopo, 2022-02-27:
Not completely out of the box, but it is possible using a custom plugin.
For example you could start off using the included "Filter" plugin. If your message matches the regular expression `.*PRIO1.*` (Anything or nothing, followed by the sequence PRIO1, followed by anything or nothing), you set a pre-defined alias. You could use a dynamic alias, but that needs additional database requests and is not that easy as using a fixed one.

You could extend filter.js by adding the following block or make a copy of filter.js, using just this block instead of the default ones:
`
    if (config.staticAliasExpression && config.staticAliasId) {
        if (data.message.match(new RegExp(config.staticAliasExpression))) {
            data.pluginData.alias_id = config.staticAliasId
            logger.main.info(`Setting alias to ${config.staticAliasId}, because message matches ${config.staticAliasExpression}`);
        }
    }
`

Additionally, you'd need to tell pagermon about the new config variables by extending filter.json (or by creating an adapted copy, if you made a copy of filter.json)
`
{
            "name": "staticAliasExpression",
            "label": "Replace Alias - Expression",
            "description": "Expression that must match, to set static alias",
            "type": "text"
},
{
            "name": "staticAliasId",
            "label": "Replace Alias - Alias ID",
            "description": "Id of the alias to use, if expression matches",
            "type": "text"
},
`

darrenjoycenz, 2022-05-24:
Hi
I've been attempting this, but nothing seems to happen to the message alias. I've configured it as a separate plugin based on Filters, and it shows in pagermon. Set the settings, but nothing happens to the messages.  I'm scratching my head and unsure where to start debugging this problem. Any advice appreciated. :)

darrenjoycenz, 2022-05-24:
fixed it.  data.pluginData.alias_id = config.staticAliasId in the post should be
data.pluginData.aliasId = config.staticAliasId

darrenjoycenz, 2022-05-24:
Just leaving this here for others that are interested.
I've implemented the above as a new plugin, but in my .js I've hardcoded the text matches.
ie

if (data.message.match(/ALPHA1/g)) { data.pluginData.aliasId = 23;}

set's the alias to number 23 if the text ALPHA1 is in the message

Hope this helps!

closed 2022-10-10 by eopo