pagermon/pagermon #506: 2 servers 1 host

AKGhost, 2022-03-12:
**Is your feature request related to a problem? Please describe.**
I am monitoring 2 Pocsag systems that share capcodes but are 2 seperate districts.

**Describe the solution you'd like**
When running the server, settings such as Hide unknown capcodes are duplicated on each server and are not independent

**Describe alternatives you've considered**
have tried renaming database and have seen 2 seperate message lists but this is not allowing seperate server instances.

**Additional context**
have been able to run default port 3000 and port 3001 for web interface but again, unable to get seperate admin panel settings.

I am not having issues with the 2 receivers as that is working with no issues at this time.

eopo, 2022-03-13:
Both your systems seem to access the same database and the same config file at the moment.
The way pagermon is currently set up does not allow two different instances running from the same codebase.
If you wish to have two separate instances, you could copy the whole pagermon directory and change the config accordingly (Most important: different port).
If you use Sqlite, that should be it. If you rely on MySQL or MariaDB, you'd need to change the database settings to use a different, second database.

We could improve pagermon in a way, that running multiple instances from one codebase could be possible. This isn't something that's gonna happen in the next few weeks tho.

AKGhost, 2022-03-13:
I see the error i made. I was running guswt the server folder under a new name but need to make a complete second server in general. Thanks for the info. 

Closing

closed 2022-03-13 by AKGhost