pagermon/pagermon #508: FontAwesome icons with hyphen not supported

andrewerrington, 2022-03-20:
**Describe the bug**
When typing the name of a Font Awesome icon in the Alias Details page, it only works for icons that do not have a hyphen in the name.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Admin|Aliases'
2. Click on 'ADD NEW'
3. Scroll down to 'Icon'
4. See initial icon is a question mark, and the name is 'question'
5. Type a new Font Awesome icon name. A good example is 'truck', which works, and 'truck-medical', which doesn't. Enjoy yourself by entering other icon names listed on Font Awesome's website.
6. Observe that icons without a hyphen are displayed. Icons with a hyphen show as blank circles.

**Expected behavior**
Any named Font Awesome icon should be displayed properly.

**Screenshots**
None available.

**Environment information:**
 - OS: v0.3.11e Pagermon Pi Edition (https://github.com/pagermon/pagermon/releases/tag/v0.3.11e)
 - Other details: Running on Raspberry Pi Zero 2 W with rtl-sdr compatible dongle.

**Additional context**
Everything seems to be working fine, apart from this one issue.

DanrwAU, 2022-03-21:
I have icons with hyphens working correctly life-ring is one I use regularly.

I suspect it is due to us using an older  version of fontawesome than the library on their website.

andrewerrington, 2022-03-21:
Hmm. I tried 'life-ring' and it worked! Could you give me some other hyphenated names that are working for you?

Could you also please try 'truck' and 'truck-medical'? Firstly, confirm at Font Awesome's site that 'truck-medical' should work (it's a truck with a + symbol on it). Thanks.

marshyonline, 2022-03-21:
Some icons cant be used because we are pined to v5 of the icon set - you can see a list of icons you can use here:
https://fontawesome.com/v5/search?m=free

There is an incompatibility issue with the latest version that we haven't been able to resolve.

andrewerrington, 2022-03-21:
Excellent! That seems to be precisely the reason, and I was able to find icons I could use in the v5 set. I'm happy that there is a root cause, and I understand why it won't be changed very soon. Thanks!

closed 2022-03-21 by andrewerrington