pagermon/pagermon #509: Utiliser 2 clés SDR

Laurent471, 2022-04-01:
Hello, is it possible to configure 2 sdr keys in the READER.SH in order to monitor 2 different frequencies?

davidmckenzie, 2022-04-01:
Sure, you could make some minor modifications to run both in the same session, would require some bashfu. Or could copy and paste the shell script and run them with something like [parallel](https://www.gnu.org/software/parallel/), or just run them in the background i.e.
```
reader1.sh &
reader2.sh
```

eopo, 2022-04-01:
Absolutely. The way pagermon is designed allows you to feed one server with unlimited clients (The name is a bit confusing, it's more like a feeder). You can host them on the same device or run them on multiple devices at multiple locations.
Best practice in the current state would most probably be to make a copy of the client folder, change the config to use a different API key previously generated in the pagermon settings (You could theoretically use the identical key, it wouldn't make a difference, but maybe future versions could have problems with that) and change the rtl_fm part of reader.sh according to your needs on the second frequency.
Please note that pagermon currently has no built in feature that can differentiate identical cap code addresses on different frequencies yet!

geelongmicrosoldering, 2022-05-23:
@Laurent471 if the above comments have helped, can you please close this issue? :smile:

DanrwAU, 2022-12-21:
Closed - Suggestions here are the way to do this

closed 2022-12-21 by DanrwAU