pagermon/pagermon #515: Documentation for (Advanced/Simple) webhook or Shell

krisken, 2022-05-19:
I'm looking for a solution to get the following:
If messages come in from certain CAP codes (several) I would like to put them in an external database.

This, for example, by calling a PHP script (on another server), where I can then send the variables, and where the script then puts them in a database.

I suspect that this is possible with a (simple / advanced) webhook or with the Shell option, but I miss the documentation to do this correctly.

Main variables I need are:
- Date / Time
- CAP code (address)
- Agency
- Alias
- Message

geelongmicrosoldering, 2022-05-19:
Yeah so enable **Simple Webhook**,
Destination URL: Your PHP file.
Filter Mode: Per Alias Only.

In each of the Alias you want, edit them and check **Enable sending messages that match this alias via Simple Webhook**

To grab the data from the array, the easiest way is probably this (one liner)
`$pagedata = json_decode(file_get_contents('php://input'), TRUE);`

That will get the contents of the file php://input (the json data that is posted to it)
It will then decode the json data that was posted into a php array. Specifically an associative array (thats what TRUE does);
The associative array is now stored in $pagedata.

You can access it using the appropriate associations

```
$lastpage = "\nDate / Time:" . $pagedata['timestamp'];
$lastpage .= "\nCAP code (address):" . $pagedata['address'];
$lastpage .= "\nAgency:" . $pagedata['agency'];
$lastpage .= "\nAlias:" . $pagedata['alias'];
$lastpage .= "\nMessage:" . $pagedata['message'];

//This will store the last page in "lastpage.txt", assuming php (well, usually www-data) has permission to write in the CWD.
file_put_contents("lastpage.txt", $lastpage);
```

Of course, you would then need to take those variables and put them into a mysql or sqlite or whatever insert statement. 

Would also be worth checking nothing is NULL/undefined, and sanitizing the input to avoid errors in data storage, something [like this](https://www.php.net/manual/en/mysqli.real-escape-string.php).

All information on taking an arbitrary PHP variable and putting it in a database though can be found on google.

Don't forget, if you are ever curious about how a post or get request is structured, and how data is sent to an endpoint, from any software that can send data to an endpoint, you can always send it to an arbitrary one to see what is getting sent, like using [Hookbin](https://hookbin.com/) etc.

geelongmicrosoldering, 2022-05-19:
For the sake of having it public, here's my reply to your message on telegram:
> In the case of sms using the url you provided, do the same, but you would do something like this:
> 
> `file_get_contents("https://smsgw.be/xxxxxx/sms.php?&from=32496xxxxxx&to=32484xxxxxx&msg=".urlencode("Alias: ".$pagedata['alias']." Message: ".$pagedata['message']));`
> 
> Which will send an sms with Alias and Message. The urlencode encodes characters for the url (a space becomes %20). 90% of sms apis will decode it for you on the other end. Otherwise, probably want to make it a post request instead.

closed 2022-06-18 by eopo