pagermon/pagermon #516: Support for european characters not working properly

Hayvosh, 2022-05-23:
The current version of multimon that is bundled does not support swedish UTF-8 characters, for instance "-C SE" from pocsag.c in multimon git repo. 
But the lastest version of multimon has this implemented and working.

geelongmicrosoldering, 2022-05-23:
@Hayvosh and I have used the following workaround to get this working, as there is decoding issues in the newer version of multimon-ng.

Basically you pipe the output of multimon via sed before reader.js, to replace specific characters that are unique to the Swedish Charset:

Swedish (SE) Charset
`multimon-ng -q -b1 -c -a POCSAG512 -a POCSAG1200 -a POCSAG2400 -f alpha -t raw /dev/stdin | sed 's/\[/Ä/g; s/\\\\/Ö/g; s/\]/Å/g; s/{/ä/g; s/|/ö/g; s/\}/å/g' | node reader.js`

German (DE) Charset
`multimon-ng -q -b1 -c -a POCSAG512 -a POCSAG1200 -a POCSAG2400 -f alpha -t raw /dev/stdin | sed 's/\[/Ä/g; s/\\\\/Ö/g; s/\]/Ü/g; s/{/ä/g; s/|/ö/g; s/\}/ü/g; s/~/ß/g' | node reader.js`

This could also be done for French and Slovenian, however there are more characters to replace.
If you are looking to get one of those working, the information on whats different about the charsets can be found in the [multimon-ng source code](https://github.com/EliasOenal/multimon-ng/blob/master/pocsag.c).

Thanks @Hayvosh , you can close this now :)

Hayvosh, 2022-05-23:
Tested and verified as @geelongmicrosoldering mentioned. Closing issue :-)

closed 2022-05-23 by Hayvosh

marshyonline, 2022-05-23:
I assue you are talking about the bundled version of Multimon with the Raspberry Pi Image.

The RPi Image has an older version of Multimon because newer versions have some new detection code that is flawed and causes pocsag messages to be missed with a high miss rate. 

You can compile and install the latest version on the pi using the standard instructions to gain this support.

Hayvosh, 2022-05-23:
> I assue you are talking about the bundled version of Multimon with the Raspberry Pi Image.
>
> The RPi Image has an older version of Multimon because newer versions have some new detection code that is [...]

You are indeed correct, this is on the Pi image.
So pagermon is using the multimon-ng file in /usr/local/bin directory then I assume?
If so, then I can try as per your suggestion later to compile the newer version and see how that works out.

geelongmicrosoldering, 2022-05-23:
@Hayvosh if you were going to compile it, id probably compile the older version and just add in the charset changes from the newer version to avoid the detection issues. You would end up with a hybrid of old/new, but it would suit your needs better. Many have enough trouble decoding without adding extra variables :stuck_out_tongue_winking_eye:

aleluja123, 2022-06-10:
How to edit to Slovenian charset?

SloCompTech, 2022-06-10:
In reader.sh you add `-C SI` to multimon-ng

SloCompTech, 2022-06-10:
You probably have too old version of multimon-ng

aleluja123, 2022-06-10:
I have installed [v0.3.11e - Pagermon Pi Edition: Electric Boogaloo](https://github.com/pagermon/pagermon/releases/tag/v0.3.11e)

How to update multimon-ng?

geelongmicrosoldering, 2022-06-10:
The included version of multimon-ng is intentionally outdated. You can use sed as above and covert the US char equivalents to SI. Im not on a computer right now but SI charset difference is in source code i linked

aleluja123, 2022-06-10:
Sorry I am not good at this...
What should I wrote in reader.sh?