pagermon/pagermon #521: Escape special characters

andrewerrington, 2022-06-14:
**Describe the bug**
If a pager message contains less than or greater than symbols the symbols are not rendered on some browsers.

**To Reproduce**
Find a message known to contain less than or greater than symbols. Observe that the symbols do not appear in some browsers. For me, the symbols are shown on desktop Firefox on Windows, but not mobile Firefox on Android.

**Expected behavior**
If a message contains less than or greater than symbols the symbols should be displayed.

**Screenshots**
No screenshots, sorry.

**Environment information:**
Firefox 101.2.0 on Android illustrates the problem.

**Additional context**
The pager message definitely contained these symbols as they correctly appeared on the pager itself. Same pager message captured by PagerMon and viewed with Mobile Firefox on two different phones does not, but does with Firefox under Windows 7.

eopo, 2022-06-18:
Has to do with the library that does the highlighting-feature. It seems to get rid of everything that looks like an html tag.

eopo, 2022-06-18:
Can you confirm the behavior described here: https://github.com/pagermon/pagermon/issues/212#issuecomment-613559811

andrewerrington, 2022-06-19:
Um. I deleted the highlighting settings (the message that showed the issue did happen to be highlighted) but the greater than and less than symbols are still not displayed on mobile Firefox. I put the highlighting back.

andrewerrington, 2022-06-20:
Ok. I now see the same problem with Desktop Firefox. Which is odd, but never mind.

I've looked in the database, and the less than and greater than symbols are present in the message content in the database itself, but they are not shown on screen. Probably need to be escaped to &lt and &gt

all3kcis, 2022-07-24:
Same here, any suggestion ?

sorenrosenberg, 2023-09-29:
It would be nice to have a proper implementation of recognising <CR><LF> as a linebreak. Also when messages are parsed on further to plugins and not just in the browser.