pagermon/pagermon #524: Migrations tool

eopo, 2022-06-25:
**Is your feature request related to a problem? Please describe.**
From time to time, new features are added to pagermon that require further configuration. In order to not break already running systems after an update, hacky bits are added to the source code.

**Describe the solution you'd like**
On startup a tool should check, if the configuration file has everything it needs to run. If not, options should be added automatically or the administrator should be asked, how he wants to set these options.

eopo, 2022-06-25:
Two proposals:
1. The config holds a variable of the pagermon version it is valid for. On startup, this version is compared against the version constant in the source code. If the configuration version is lower than the current version, the migration tool tries to find and run a migration script for every version between the config version and the current version. On a completely fresh system without a config, a initialization migration is run, that creates the config and interactively asks the administrator for values via a cli.
1.1. On a fresh system, the initialization migration provides a config that includes everything that is needed for the current version.
1.2. On a fresh system all migrations will be run (Similar to how knex migrations work. All systems, wether they upgrade versions progressively, are freshly set up or upgrade from a few versions back to current version undergo exactly the same steps, so the result should be exactly the same).
2. Similar like its currently done within the main app code, migrations check for the value or existence of configuration items by a premise or assertion. If they return false, the migration script is run.
```JS
{
    assertion: function(config, database){
        // Check, if configuration contains a database port
        try {
            return typeof config.get('database:port') === "Number"
        } catch (e) {
            return false
        }
    },
    migration: function(config, database){
        // Add standard port of 3306 to configuration
        config.set('database:port', '3306')
    }
}
```

What do you like best, or do you have additional ideas?

geelongmicrosoldering, 2022-06-25:
I definitely think the latter is a better approach.
Having a heap of migration scripts, or relying on checking a version number to determine if a value may or may not be there, doesn't really make sense. Its a whole bunch of extra code that really only works if you include the assertions/expectations anyway.

The version performing the check knows what data it expects. It doesnt need to know what version it is, what version its coming from, or what should be (but also may not be) in between.

So, if the version running doesnt expect a port number, it doesnt check for it, and doesnt add it.

If it does expect a port number, it looks for it and either uses it, or sets a default and uses that.

This also means that if someone copies out a block from an old config (say, the database config, without changing the version number), the default port is still applied without it being skipped because of the assertion that "the config file is new enough, so it must be ok".

I think the "we need this value, we dont have this value, use the default value" is an easy enough approach. Its easy for PRs that require config file changes, and it doesnt require terminal interaction.

Id just drop out the current app.js method to a seperate file to make it easier to find.

Worth pointing out fyi, that database port can be both a number, 3306, or a string, "3306". In your example above, '3306' probably wouldn't be a number.

When i tested it and provided an example in that PR, i had it set the default in app.js as a number because it was 2 less buttons to press. But if you save it in admin panel, it turns it into a string.

Both work, but checking if its a number might actually just overwrite the custom port on startup every time. I guess you could do a regex match for [0-9]+ or something. That conversation isnt really relevant to this PR though. 😜