pagermon/pagermon #525: Improve terminology

eopo, 2022-06-25:
**Is your feature request related to a problem? Please describe.**
Both in the source code and in the documentation, some terms are used ambiguously, while other things are described with multiple terms. Examples would be "message", which refers to both a telegram and its content. The object that links a RIC with information about its representation is called both "capcode" and "alias", although both terms have different meanings.

**Describe the solution you'd like**
We should disambiguate the terms while staying as close as possible to the specification of the respective protocols. Problems arise because POCSAG and FLEX sometimes use different terminology for identical properties. In Europe, the term "RIC" is commonly used under POCSAG to designate the address of a pager, although "cap code" is also used in the standard. A consensus must therefore be found here.

geelongmicrosoldering, 2022-06-25:
I've noticed some things like this. Terms that dont align. It happens in code too. Actually i noticed the other day in the message api, the key for epoch time could be "timestamp" or "datetime" depending on if its a get request, or a post.

eopo, 2022-06-25:
Yeah, just things that happen when many people work on many parts of code over a long period of time.
It will be a breaking change, but I guess it is worth it. We should make some definitions of what's called in the Documentation and everything added should adhere to that.