pagermon/pagermon #527: Client sends to multiple servers

andrewerrington, 2022-07-04:
**Is your feature request related to a problem? Please describe.**
I have PagerMon working great on a single computer. Messages come in to the client from a TV dongle and are sent to the server on the same machine. The messages are displayed by the webserver on the machine too.

I'd like to be able to see the messages elsewhere, but I can't expose the webserver to the outside world.

**Describe the solution you'd like**
I realise that a single PagerMon server can accept multiple message streams from different clients, but it seems that the client can only be configured to send data to a single server. I'd like it to be possible to have a list of several servers in the client configuration, and for the client to be able to send to all of them when a message arrives. In most cases the list will have one entry and PagerMon will behave as it currently does.

**Describe alternatives you've considered**
I've tried nothing, and I'm all out of ideas. There may be some Unix wizardry that lets me duplicate a data stream and send it to another host, or a way to replicate an sqlite database elsewhere, but I haven't looked into it.

**Additional context**
No additional context, sorry.

geelongmicrosoldering, 2022-07-04:
Assuming you just need one additional location, you can use message repeat plugin. You'll find it in admin settings.  You can daisy chain servers that way too. server 1 > 2 > 3 > 4

andrewerrington, 2022-07-04:
I think that is exactly what I want. Thanks!

closed 2022-07-17 by eopo