pagermon/pagermon #528: Migration from SQLITE to MYSQL

all3kcis, 2022-07-24:
**Is your feature request related to a problem? Please describe.**
I want to migrate my actual DB (SQLITE) approx 130k row in messages to MYSQL DB, but all the tools I have tested report errors

**Describe the solution you'd like**
I didn't find it in the wiki but maybe you have something planned for that already?

DanrwAU, 2022-07-24:
Export to tsv from sqlite then import into mysql is how I do this.

all3kcis, 2022-07-24:
I exported in CSV, I met problems with the field alias_id of the table message, in CSV the column was empty and MYSQL waited a value "NULL" in this case, I could manage by modifying these lines with Excel.

To recreate the tables I configured pagermon and then restarted it. But it doesn't recreate the "messages_search..." tables, is that a problem?

eopo, 2022-07-26:
The best way would be to change config to mysql and then run pagermon. This will start the database initialization process. After that, shut down pagermon, export your SQLITE-tables for messages and aliases to SQL insert statements. First import your aliases, then your messages. If you try it the other way around you will get reference errors.

BuildTheRobots, 2022-08-18:
FYI: I found exporting from SQLITE to TSV was causing the data to get corrupted when opening in LibraOffice.
I ended up exporting using a custom separation character (®) and then writing a tiny perl script to convert each line to a CURL command which uses the API to insert the messages into the new database, which seems to be taking care of aliases and search indexes for me, without me needing to understand the SQL data structure.
180,000 messages; it's been going since last night and seems to be working as expected; just somewhat slow.

eopo, 2022-10-12:
That's certainly another way to do it.
I'm gonna close this issue for now. Maybe we'll offer a migration tool in the future, but that's something further down the road.

closed 2022-10-12 by eopo

RIKIKU, 2022-10-28:
I built something to do this in PR #462
There was suggestion from someone in that thread to use https://github.com/techouse/sqlite3-to-mysql also