pagermon/pagermon #533: Version for armv8 (aarch64)

Olivier6767, 2022-10-30:
Would it be possible to make pagermon compatible with aarch64 to run on Rpi 4 ?
I switched my raspberry that holds all the RTLSDR sticks to Ubuntu 22.04 64 bits and performance is improved.
I tried running pagermon on it but  I get no Pocsag messages.
I guess there must be an incompatibility since there is a specific armv7 version in addition to the standard one.

marshyonline, 2022-10-30:
Let me look into this 👍

marshyonline, 2022-10-30:
Are you talking about the Pre Compiled Pi Images? - They are compiled against armhf
I can look at building a AArch64 image of rasbian - I don't think it needs anything special

Olivier6767, 2022-10-30:
Yes, this would be great!

leecollingsco, 2023-07-31:
Hi, does this mean as I'm trying to install this on my RPi4 (64bit) right now, it's not going to work?