pagermon/pagermon #544: Please improve the Installation Guide for less savvy users

Swekkerooni, 2022-12-29:
Hello,

I have spent considerable time of this Christmas break to get PagerMon installed on my home server, which runs a couple different software packages, most of which are in Docker. I use docker-compose.

The guide that is given in the readme seems complete, but is missing a lot of information for less savvy users.

1. PagerMon does not work in Safari. Period. This was extremely frustrating/infuriating, as I spent 3 whole days setting up the Docker server multiple times over and over, wanting to generate the API keys. But when accessing the Pagermon web interface on IP:3000, it partially loads some page elements but it just overlays a spinning icon. Using Chrome, it worked immediately.  It would be recommended to add this as a note for instance in the readme in the section about getting the API-keys, as it is probably the first time a user would access the webinterface.
2. The docker install guide for the pagermon server needs some improvement to make it more clear. Just start out with the docker-compose guide, as it is probably the most user-friendly install method. Building pagermon, whomever would be interested in that, is perhaps better placed somewhere else, instead of the being the first suggestion in the guide.
**2.a** Same with the `:<VERSION> `stuff, just please take that stuff out! Omitting `:<VERSION> `just pulls the latest version right??? That is what you want for 99% of the users right??? I tried appending the current version in my first iteration but it couldn't find with `:0.2.13a`. Just removing the `:<VERSION>` fixed the errors I had and at least installed _something._
3. Installing the pagermon client is super unclear in relation to both docker OR the local installation method. Like, what do you do after editing the config.json?? If it should magically spring to life, it doesn't. If that should be the case nonetheless, maybe write what the expected behavior should be. Because right now the guide ends with editing the config.json, giving the impression everything should work together with the pagermon server after that. I have done plenty of system reboots in before, between, during and after.
**3.a** _If_ the client **is** installed together with the docker pagermon-server, this is written nowhere. If that is the case, where do I set-up the arguments for rtl_fm and multimon-ng?
4. Getting the pagermon client installed is a crapshoot in the first place as well, because, if you install the server using docker-compose, it creates a folder named `pagermon`. The problem arises when you want to run `git clone https://github.com/pagermon/pagermon.git`, as it sees that the directory `pagermon` already exists.
5. While this can be fixed by pulling into another directory, this took me 8 hours to find out. After that I did everything as stated in the guide, but no successes, again, probably because the client is expecting to work with the npm-installed pagermon server located in its' parent directory. Except I have installed with docker!

After that I tried to troubleshoot everything, running the correct `rtl_fm | multimon-ng` command from terminal just to get some feedback. Then I found out multimon-ng wasn't even installed? I've never used `pm2` or `npm` before so I'm totally unsure if multimon-ng should have been installed by those running `npm install`.

DanrwAU, 2022-12-29:
Most of your issues seem to be docker related.

Unfortunately the app wasn't written with docker in mind and can be tricky. The guide in the wiki is 95% there for running natively without docker.
Personally I wouldn't ever recommend running it in docker. 

Feel free to write some docs and submit a pull request, when you've used a project for so long (and contributed) it can be hard to see the gaps for those of us regulars. 

Feel free to hit up #support on the discord even if nobody is active chances are questions have been answered there.

Swekkerooni, 2022-12-29:
I would love to write docs for Pagermon, as I'm a technical copywriter for my job. I've got a knack for idiot-proofing guides like these, because, I am an idiot myself. But in order to do that, I need to get it working for myself in the first place.

I forgot to mention that I also tried the local setup method a couple times but whatever I do, I can't get the client installed correctly, or it is unclear what should be done after editing the config.json of the client.

Edit: For now I'm back to square one though, because whatever I do running `rtl_fm -f 169.65M -M fm -s 22050 | multimon-ng -a FLEX -t raw -`, it just gives garbled output or no output at all.

geelongmicrosoldering, 2022-12-29:
There is a step missing there that i noticed today is only mentioned in the raspberry pi instructions.

After editing the config, from the client directory, run the command 'pm2 start reader.sh'
You can then run `pm2 save` and `pm2 startup` which will allow the running server/client to survive reboot.

But the issue you are facing needs to be solved first. Unfortunately its going to be an issue with rtl-fm config, multimon-ng, or the receiving hardware/antenna itself.

You might be best to start with rtl-fm and pipe the audio somewhere so you can listeb

geelongmicrosoldering, 2022-12-29:
Sorry, i pressed the wrong button there. Didnt mean to close your issue :)