pagermon/pagermon #545: Wildcard

112bzo, 2023-01-03:
Hello.

I like to use a wilcard. 

my capcode 00117.... to 00118.... like to see. So i like to see all caps in the range of 117 to the end of 118

How to do that

I have try 00117____ and also 00118____ but i dont see the range

geelongmicrosoldering, 2023-01-03:
How certain are you about those lengths?

00117____ is a 9 digit capcode. If this is pocsag, you probably have an extra __ there.

All of my pocsag addresses are 7 digit (or less, preceded by 0, which is how multimon-ng interpets shorter addresses.

The fact that yours start with 00, makes me think they are 5 digits. Which would mean the correct wildcard match would be `00117__` with would match 0011700-0011799. You would need a second alias, `00118__` to match the rest from 0011800-0011899.

Double check those address lengths. The exact number of digits are super important.

112bzo, 2023-01-04:
Thats true, i use flex protocol, and i must use 00 before te real capcode. 

But i works, only 00118000 is not what i whant, i like the range 00118 but whitout 00118000 is that posible.

Start ranch imust be 00118001 to 00118999 so how to do that

Sorry for my bad language.

geelongmicrosoldering, 2023-01-04:
Well now you have 8 digits. The nunber of digits is very important, as `_` corresponds to exactly 1 character.

To get the range of 00118001 - 00118999:
1. Create an alias, using `00118___`. Give it a name like "Important Pages".
2. Create another alias using `00118000` and check the "Filter Alias" (Ignore messages matching this alias.) box. Give it a name like "Junk".

1 will get you everything including the address you dont want.
2 will filter out the address you dont want

closed 2024-11-01 by eopo