pagermon/pagermon #546: node:internal/readline/emitKeypressEvents:71 error while monitoring a busy FLEX paging channel.

KD7DMP, 2023-01-05:
I've been running Pagermon for around 2 months now to monitor a Paging system my Company helps to support. It's been a great tool to keep track of what is going out on the system and to look up CapCodes for customers that are missing pagers.

I just recently added a second channel to the system for a new customer of ours and noticed the original channel intermittently having this error. 

`Jan 05 11:38:37 Radio reader.sh[9830]: 2023-01-05 11:38:37: 000704866: THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  2419
Jan 05 11:38:37 Radio reader.sh[9830]: 2023-01-05 11:38:37: 000602836: Q>]6M3a3`f<amM1=mK<?GLAK?dl7OefS0]WDToOeFZ9
Jan 05 11:39:19 Radio reader.sh[9830]: 2023-01-05 11:39:19: 000603497: [9=7eT0EgQ=gE1m1F=lmWE0C\WaIU8Fg2kK@SfKL]`oEdho8;gf;>l5HlLZC<fek3MUlZHDa?2Da@9P\Mc20`_YM=lOKgCDQ?Cm0]Yl;b9=
Jan 05 11:39:19 Radio reader.sh[9830]: 2023-01-05 11:39:19: 000830538: THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  2420
Jan 05 11:39:24 Radio reader.sh[9830]: 2023-01-05 11:39:24: 000603084: 777-7777
Jan 05 11:39:49 Radio reader.sh[9830]: 2023-01-05 11:39:49: 000603097: 777-7777
Jan 05 11:39:49 Radio reader.sh[9830]: 2023-01-05 11:39:49: 000603084: 777-7777
Jan 05 11:39:55 Radio reader.sh[9830]: node:internal/readline/emitKeypressEvents:71
Jan 05 11:39:55 Radio reader.sh[9830]:             throw err;
Jan 05 11:39:55 Radio reader.sh[9830]:             ^
Jan 05 11:39:55 Radio reader.sh[9830]: TypeError: Cannot read properties of null (reading '1')
Jan 05 11:39:55 Radio reader.sh[9830]:     at Interface.<anonymous> (/root/pagermon/client/reader.js:121:77)
Jan 05 11:39:55 Radio reader.sh[9830]:     at Interface.emit (node:events:513:28)
Jan 05 11:39:55 Radio reader.sh[9830]:     at Interface._onLine (node:readline:491:10)
Jan 05 11:39:55 Radio reader.sh[9830]:     at Interface._line (node:readline:868:8)
Jan 05 11:39:55 Radio reader.sh[9830]:     at Interface._ttyWrite (node:readline:1227:16)
Jan 05 11:39:55 Radio reader.sh[9830]:     at Socket.onkeypress (node:readline:291:10)
Jan 05 11:39:55 Radio reader.sh[9830]:     at Socket.emit (node:events:513:28)
Jan 05 11:39:55 Radio reader.sh[9830]:     at emitKeys (node:internal/readline/utils:357:14)
Jan 05 11:39:55 Radio reader.sh[9830]:     at emitKeys.next (<anonymous>)
Jan 05 11:39:55 Radio reader.sh[9830]:     at Socket.onData (node:internal/readline/emitKeypressEvents:61:36)
`
But, it's only having this error on the original system. The Original system is a mix of POCSAG 1200 and FLEX, but the new system is only POCSAG. So I'm thinking this may be an issue with FLEX decodes.

Another thing that might be causing this is the channel that is having the error carries encrypted FLEX traffic, So maybe it's having an issue parsing that?

*additional note*
Looking back in the logs it likes this has been happening for a while now, but I just didn't notice it because I would restart the service it's attached to every weekend when I did server backups. We haven't had many issues with the system for the last month so I guess I didn't notice the missing pages in the database. :-/

Any help on this would be appreciated. If this program works out in the long run I'm looking at deploying it on a raspberry Pi and using it as portable coverage monitor for customer reporting system issues in one area or who are on the edges of the system. 
Thanks for the great application and all the hard work on it!

DanrwAU, 2023-01-06:
Line 121 in reader.js relates to FLEX so i think your correct there. 

I don't think this is actually a pagermon issue, looks like the upstream dependancies are having issues parsing *something*  but you'd need to log the output directly to figure it out, whatever is causing the issue isn't in these logs based on the timestamp it is receiving some kind of message that is crashing it out. 
If you can get a log of all the messages received that might help

geelongmicrosoldering, 2023-01-06:
+1 for logging output directly to get more information. You can pipe the output to a file before reader.js `rtl_fm ... | tee messages.txt | node reader.js`
That becomes problematic if its happening once a week though, file could get very large.

You can also mitigate the restart issue by just having it restart itself after 10 seconds in a while loop.
```
while true; do
rtl_fm ..... reader.js
sleep 10;
done
```

That way if you dont work out specifically whats causing it, at least it corrects itself

KD7DMP, 2023-01-06:
@geelongmicrosoldering Thanks for the input. I don't think this will be too much of a problem today. I've seen it crash 5 times over the last 2 hours. So what ever pager is causing it to dump is going out quite a lot today.  
I setup your idea with tee and I'll see what I get.

KD7DMP, 2023-01-06:
Sorry, I didn't mean to close this thread. I'm still new to GitHub. :-/

geelongmicrosoldering, 2023-01-06:
Thats cool, last time i did it by pushing the button accidentally

KD7DMP, 2023-01-06:
OK, so it looks like what might be causing the problem is a "blank" alpha
page in Flex.
POCSAG1200: Address:  201666  Function: 2  Alpha:   1444<ETB>
FLEX: 2023-01-06 00:51:42 3200/4/K/A 13.000 [000602836] ALN
Q>]6M3a3`f<amM1=mK<?GTA2I_m50CLBDSfNfagRZU<
FLEX: 2023-01-06 00:51:42 3200/4/K/C 13.000 [000830538] ALN THIS IS A TEST
PERIODIC PAGE SEQUENTIAL NUMBER  2732
FLEX: 2023-01-06 00:52:21 3200/4/K/A 13.020 [000602836] ALN
Q>]6M3a3`f<amM1=mK<?G<<D[kPWkc9XA`2VI1UNBk9
FLEX: 2023-01-06 00:52:21 3200/4/K/C 13.020 [000830538] ALN THIS IS A TEST
PERIODIC PAGE SEQUENTIAL NUMBER  2733
FLEX: 2023-01-06 00:53:35 3200/4/K/A 13.060 [000602836] ALN
Q>]6M3a3`f<amM1=mK<?GlC<U=3jRjRQGbWBf><dlc7
FLEX: 2023-01-06 00:53:35 3200/4/K/C 13.060 [000830538] ALN THIS IS A TEST
PERIODIC PAGE SEQUENTIAL NUMBER  2734
FLEX: 2023-01-06 00:54:12 3200/4/A 13.080 [000603097] NUM 777-7777
FLEX: 2023-01-06 00:54:15 3200/4/K/A 13.081 [000602836] ALN
***@***.***[oc:Pi5
FLEX: 2023-01-06 00:54:15 3200/4/K/C 13.081 [000603498] ALN
XYl91mNePJ]Jcli>i6Uej3
FLEX: 2023-01-06 00:55:28 3200/4/A 13.120 [000603084] NUM 777-7777
FLEX: 2023-01-06 00:55:28 3200/4/A 13.120 [000603097] NUM 777-7777
FLEX: 2023-01-06 00:55:28 3200/4/K/A 13.120 [000603622] ALN
***@***.***
***@***.******@***.***
***@***.***[M`1CaFGPO5YH^YN
***@***.***>
FLEX: 2023-01-06 00:55:28 3200/4/K/C 13.120 [000704866] ALN THIS IS A TEST
PERIODIC PAGE SEQUENTIAL NUMBER  2736
FLEX: 2023-01-06 00:56:08 3200/4/K/A 14.013 [000603257] ALN
***@***.***
Subj:WHF MAIN OR POOL STAT Inpatient consult to IV team Sent
Msg:Epic In Basket: Orders Message: ________________________________ This
message is intended for the sole use of the addressee, and may cont
FLEX: 2023-01-06 00:56:08 3200/4/K/C 14.013 [000603256] ALN
***@***.***
Subj:WHF MAIN OR POOL STAT Inpatient consult to IV team Sent
Msg:Epic In Basket: Orders Message: ________________________________ This
message is intended for the sole use of the addressee, and may cont
FLEX: 2023-01-06 00:56:42 3200/4/K/A 14.032 [000602836] ALN
Q>]6M3a3`f<amM1=mK<?GPDNnWnl4h1IE<khhPFWS:8
FLEX: 2023-01-06 00:56:42 3200/4/K/C 14.032 [000830538] ALN THIS IS A TEST
PERIODIC PAGE SEQUENTIAL NUMBER  2737
FLEX: 2023-01-06 00:56:48 3200/4/A 14.035 [000603084] NUM 777-7777
FLEX: 2023-01-06 00:56:48 3200/4/F/B 14.035 [000389393] ALN

The EOF is where the error took place. I used the Tee command idea
@geelongmicrosoldering <https://github.com/geelongmicrosoldering>
suggested.

geelongmicrosoldering, 2023-01-06:
Interesting. There was a PR somewhere with a fix for zero length POCSAG pages. It was never included because there was other parts to it that remained unnecessary or unfinished. I cant recall. Let me see if i can find it and pull that part out and make it work for both FLEX and POCSAG

geelongmicrosoldering, 2023-01-06:
Ok i can see an issue with Line 121, group 1 in the regex match is always assumed to be there. when the ALN page is blank, it isnt, so assuming it is is the issue.

I'll get back to you with a potential fix

geelongmicrosoldering, 2023-01-06:
on line 121 in reader.js
can you please replace:
`message = line.match(/FLEX[:|].*[|\[][0-9 ]*[|\]] ?...[ |](.+)/)[1].trim();`
with
`message = line.match(/FLEX[:|].*[|\[][0-9 ]*[|\]] ?...[ |]{0,1}(.*)/)[1].trim();`

the regex change basically means that group [1] always exists, but can be null/0 length. this will later be caught on line 163 if it is empty, if you can also replace on line 163:
`if (address.length > 2 && message) {`
with
`if (address.length > 2 && trimMessage.length > 1)`
this will prevent the message from being sent to the API if it is empty (which will cause reader.js to crash again, albeit in a different place. In your case though, that blank message is actually fragmented. theres probably more to it.

If you can make those 2 changes and restart reader.sh, let me know if that fixes your issue. if it does i'll submit a PR to have it fixed in the next release

KD7DMP, 2023-01-06:
Done.
I'll let you know what I come up with.
I'll also do some digging on that CapCode and see if it's sending a batched
message or something. On that system it's set to a max length of 256
characters per page, so I can imagine it's breaking the page up between
batches. Unless someone is sending a whole damn email or something silly
like that.

Thank you very much for the quick response!

geelongmicrosoldering, 2023-01-06:
yeah not sure, im not familiar enough with FLEX, but based on code others have written
`3200/4/K/C` in the previous page makes it a complete page
`3200/4/F/B` in the offending page makes it a fragment, so you'd expect there to be more after that

KD7DMP, 2023-01-06:
So it looks like your change to the code at Line 121 and 163 did the trick
for the empty pages. Here is a big block of them being sent to a group of
different cap codes and pagermon handled with no issues.

LEX: 2023-01-06 03:17:20 3200/4/K/C 04.052 [000830538] ALN THIS IS A TEST
PERIODIC PAGE SEQUENTIAL NUMBER  2878
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000606716] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [001529398] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [001529398] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000683966] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000677434] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000923615] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000760621] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000957243] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000989151] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [001531192] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000728364] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000777514] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000989995] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000782259] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [000973098] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [001529530] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [001140447] ALN
FLEX: 2023-01-06 03:17:20 3200/4/F/D 04.052 [002008008] ALN
FLEX: 2023-01-06 03:18:36 3200/4/K/A 04.092 [000602836] ALN
Q>]6M3a3`f<amM1=mK<?GH=j_eIYVD69QJe5KN^V\43
FLEX: 2023-01-06 03:18:36 3200/4/K/C 04.092 [000830538] ALN THIS IS A TEST
PERIODIC PAGE SEQUENTIAL NUMBER  2879

But now we have a different error coming up.

node:internal/readline/emitKeypressEvents:71
           throw err;
           ^

TypeError: Cannot read properties of undefined (reading 'length')
   at Interface.<anonymous> (/root/pagermon/client/reader.js:163:42)
   at Interface.emit (node:events:513:28)
   at Interface._onLine (node:readline:491:10)
   at Interface._line (node:readline:868:8)
   at Interface._ttyWrite (node:readline:1227:16)
   at Socket.onkeypress (node:readline:291:10)
   at Socket.emit (node:events:513:28)
   at emitKeys (node:internal/readline/utils:357:14)
   at emitKeys.next (<anonymous>)
   at Socket.onData (node:internal/readline/emitKeypressEvents:61:36)
Signal caught, exiting!

This was the block of pages the system sent out right before that error.

FLEX: 2023-01-06 16:26:44 3200/4/B 07.097 [001372166] TON
FLEX: 2023-01-06 16:26:44 3200/4/B 07.097 [002030734] NUM 56852
4236135-5686U45]09-U916 7 4 4U4929[
FLEX: 2023-01-06 16:26:44 3200/4/B 09.097 [000988978] NUM
00000000000000000000000000000000000000000
FLEX: 2023-01-06 16:26:44 3200/4/B 09.097 [000494088] NUM 0000000000
<http://voice.google.com/calls?a=nc,%2B10000000000>
FLEX: 2023-01-06 16:26:44 3200/4/B 09.097 [001187643] NUM
0000000000000000000000000000000
FLEX: 2023-01-06 16:26:44 3200/4/B 09.097 [000920470] NUM   2 9 0684U ]9[
93221648- 05-42
FLEX: 2023-01-06 16:26:44 3200/4/B 09.097 [001169056] NUM  03-3525 5 49
745234-  2 9 0684
FLEX: 2023-01-06 16:26:44 3200/4/B 09.097 [001085614] NUM 0-69  12U733
4[494969252[159 053U 0723812
FLEX: 2023-01-06 16:26:44 3200/4/C 09.097 [000603084] NUM 777-7777

geelongmicrosoldering, 2023-01-06:
Based on what i can see there, that last message should have passed 777-7777 into trimMessage, so its length should have been defined. I havent checked, it's possible the trim function is removing the 0000's in the messages just before it.

I can see a potential issue with the code i provided.

Can you replace line 163 with:
`if (address.length > 2 && message && trimMessage.length > 1)`

And see if that helps? Im tired so i may not be processing that error to the best of my abilities lol. If it doesn't help ill take another look later today

KD7DMP, 2023-01-06:
I don't think it's the 777-7777 message either. I restarted pagermon this morning after I saw it crashed and I've gotten several of those pages with no problem. This is one from just a short bit ago.

```
2023-01-06 11:53:37: 000601338: From:EpicAllOTXSUPPORT@providence.org____ This message is intended for the sole use of th
2023-01-06 11:53:41: 0201886: .    .       .
2023-01-06 11:54:13: 000603084: 777-7777
2023-01-06 11:54:13: 000603097: 777-7777
2023-01-06 11:54:15: 000602836: Q>]6M3a3`f<amM1=mK<?G@nF9W8fhZ_MVRJ[SI<Z@F=
2023-01-06 11:54:18: 000704866: THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  3875
```
I haven't seen any of the mulit 0 messages again, so it may have been that. 

I'll keep an eye on it while I work on other projects today(Luckily I should be in the office all day today) and see if it happens again. If it does I'll implement your change and see if that helps.

Thanks again @geelongmicrosoldering for your help. It's very much appreciated.

KD7DMP, 2023-01-06:
OK, I just confirmed it's the pages with all the 0 in it. 

```
FLEX: 2023-01-06 21:28:36 3200/4/F/B 08.028 [000844961] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/B 08.028 [000576809] NUM [[700000[[[[[10000 [[[[700000[[[[[10000
FLEX: 2023-01-06 21:28:36 3200/4/F/B 08.028 [000381378] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/B 08.028 [001018284] NUM [[700000[[[[[10000 [[[[700000[[[[[10000
FLEX: 2023-01-06 21:28:36 3200/4/F/B 08.028 [000282407] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/B 08.028 [001182494] NUM [[700000[[[[[10000 [[[[700000[[[[[10000
FLEX: 2023-01-06 21:28:36 3200/4/K/C 08.028 [000602836] ALN Q>]6M3a3`f<amM1=mK<?Gd7MoEOU<F1B`WbNY\B7:W8
FLEX: 2023-01-06 21:28:36 3200/4/C 08.028 [000603097] NUM 777-7777  
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [000079176] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [002057931] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [001038158] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [000795810] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [001216332] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [001068191] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [000788766] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [000270670] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [001024436] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [001150392] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [000919602] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [001070384] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [000976697] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [000447519] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [000545585] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [001107005] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [002007976] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [000008739] NUM 000000000000000000000000000000000000000
FLEX: 2023-01-06 21:28:36 3200/4/C/D 08.028 [000389393] ALN 
FLEX: 2023-01-06 21:28:36 3200/4/D 08.028 [001916877] NUM 000000000000000000000000000000000000000
```
That was the last batch of pages I have in my Tee log file from Multimon-NG before I got the most recent version of the error message. 

I'll implement your change @geelongmicrosoldering and let you all know what I come up with.

geelongmicrosoldering, 2023-01-07:
i was able to reproduce that error locally, and can confirm the last change to line 163 fixed it

KD7DMP, 2023-01-09:
Just a quick update on this problem.
It ran without issues up till about Saturday afternoon when the process was
killed for some reason. I'm not sure why(It wasn't a fault with Pagermon).
So I restarted it and will keep an eye on it today while I'm working.
So far the fix at line 163 seems to have fixed the issue, but I've haven't
seen an all 0 pagers in the Tee file yet. Once I see one of those go
through I'll call it good.

Message ID: <pagermon/pagermon/issues/546/1374354526
> ***@***.***>
>

KD7DMP, 2023-01-09:
So I have not seen an all 0 page yet today. So I sent one to our tech test pager and it went through to pagermon with out an issue. 

```
FLEX: 2023-01-09 21:16:06 3200/4/K/C 04.012 [000704866] ALN THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  8277
FLEX: 2023-01-09 21:16:07 3200/4/C/A 04.013 [000604200] ALN `GCW6CXejh4^:gMG[h?GEB4VBf@NG7VH^i4nh9C8TdhcYFj1
FLEX: 2023-01-09 21:16:42 3200/4/K/A 04.032 [000602836] ALN Q>]6M3a3`f<amM1=mK<?GPY0j<\o4EhJhiKGj6GX=Q1
FLEX: 2023-01-09 21:16:42 3200/4/C 04.032 [000603084] NUM 777-7777  
FLEX: 2023-01-09 21:17:20 3200/4/A 04.052 [000603097] NUM 777-7777  
FLEX: 2023-01-09 21:17:20 3200/4/K/A 04.052 [000830538] ALN THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  8278
FLEX: 2023-01-09 21:17:57 3200/4/K/A 04.072 [000120065] ALN 000000000000000000000000000000
FLEX: 2023-01-09 21:18:03 3200/4/K/A 04.075 [000704866] ALN THIS IS A TEST PERIODIC PAGE SEQUENTIAL NUMBER  8279
FLEX: 2023-01-09 21:18:09 3200/4/A 04.078 [000603084] NUM 777-7777  
```

So I'm going to mark this as closed and if it comes back up I'll open this bug back up.

Many thanks to @geelongmicrosoldering for all your help on this. If I use this for our remote monitoring project at work I'll let you all know.

closed 2023-01-09 by KD7DMP