pagermon/pagermon #550: Cant search for words that have a - in it

svenskoni, 2023-03-20:
**Describe the bug**
Hello Friends,
I have the problem that when I search for the word "Intern-1" I only get alarms with the word "Intern-2" and no alarms with Internal-1 are displayed to me . This concerns all alarms with a -. Some things I can't search for at all, such as "Verlegung-1 ohne Sonderrechte".

**To Reproduce**
1. Go to 'search an type anything with a -'

**Expected behavior**
If i search for "Intern-1" it should show only "Intern-1" and not "Intern-2"
**Screenshots**
[image]

**Environment information:**
 - OS: Ubuntu 22.04.2 LTS
 - Other details: Pagermon 0.3.12-beta

geelongmicrosoldering, 2023-03-20:
Try searching for `Intern%1`, does that work?

Dev note for later:
Possibly related to Boolean matches/hyphens being a modifier. Needs further investigation oneday
https://github.com/pagermon/pagermon/blob/a5c983d7d1445cf10ea3e2aa3ad77dae11049476/server/routes/api.js#L599

svenskoni, 2023-03-20:
Nope didnt work 
[image]

geelongmicrosoldering, 2023-03-20:
It did in part. It got rid of the "but not if it contains 1" condition the -1 was causing. Intern-1 translates to "Contains Intern, but does not contain 1".  Im busy for a couple of days but if nobody else does in the meantime I'll look into some kind of workaround

DanrwAU, 2023-06-30:
The search code is terrible since i changed us over to Knex. I have nfi how to fix it though 🤣 There was an enhancement in somewhere to improve/expand on the search functioinality.