pagermon/pagermon #551: Cant Display German ÄÖÜß

svenskoni, 2023-03-20:
**Describe the bug**
Pagermon reader.js malformed the German Umlaute äöü ÄÖÜß

**To Reproduce**
Getting an Alarm with one of the Letter and you get some cryptic other Letter

**Expected behavior**
A clear and concise description of what you expected to happen.
It Should Display the Right Letter

**Screenshots**
Should be Kölner Straße
[image]

Should be Hürth Gleuel Burgstraße
[image]

**Environment information:**
OS: Ubuntu 22.04.2 LTS
Other details: Pagermon 0.3.12-beta

**Additional context**
i already have a Fix for this Problem. We need to change in Line 97 and 100 in reader.js 
97
`            trimMessage = message.replace(/<[A-Za-z]{3}>/g, '').replace(/\[/g, "ä").replace(/\]/g, "ü").replace(/\\/g, "ö").replace(/\~/g, "Ä").replace(/\!/g, "Ü").replace(/\@/g, "Ö").replace(/\#/g, "ß").trim();`
100
`            trimMessage = message.replace(/<[A-Za-z]{3}>/g, '').replace(/\[/g, "ä").replace(/\]/g, "ü").replace(/\\/g, "ö").replace(/\~/g, "Ä").replace(/\!/g, "Ü").replace(/\@/g, "Ö").replace(/\#/g, "ß")`

geelongmicrosoldering, 2023-03-20:
This issue is actually with the charset multimon-ng is using in your reader.sh

You could definitely use the replace function in reader.js, or pipe everything through sed, etc. But technically the appropriate fix is at the source.
https://manpages.debian.org/testing/multimon-ng/multimon-ng.1.en.html#C

You would want `-C DE`

Of course it really depends on your environment. You may need to update multimon-ng

A few people have had this issue with German/Swedish pocsag.

We wouldn't hard code your fix, because it might break some US encoding ect

DanrwAU, 2023-06-30:
This is a wontfix for us. You need to select the correct Charset in Multimon as per @geelongmicrosoldering, different versions of multimon also handle this differently so makes it difficult to predict.

eopo, 2023-08-29:
Well, not quite - We have hardcoded reader.js to replace the German charset to us charset. Since multimon now has the -C flag, we probably should remove that from reader.js and tell people to use the right Charset in Multimon.
@svenskoni Lösung: In Zeilen 97 und 100 der Datei reader.js befindet sich Code, der Umlaute entfernt und durch die Englische Codierung ersetzt. Wenn du den einfach löschst, sollte es gehen. /// Remove parts of lines 97 and 100 of reader.js.

geelongmicrosoldering, 2023-08-30:
> Well, not quite - We have hardcoded reader.js to replace the German charset to us charset. Since multimon now has the -C flag, we probably should remove that from reader.js and tell people to use [...]

I forgot that was there. It probably shouldn't be. Now I'm remembering that we are using an outdated multimon-ng in rpi builds for... Reasons? I feel like @DanrwAU or @marshyonline know the answer, less consistent decoding in newer builds or something.

So on that basis we probably need to either update multimon-ng or have some charset translations that can be changed in config or something.

For now i feel like those replace functions shouldn't be there, seeing as umlauts etc shouldn't come out of multimon anyway unless you are for whatever reason using a multimon-ng build that is configured to use that charset, but expecting english output. So i feel like those replace functions rectify a rare edge case of misconfiguration and it shouldn't be hardcoded to replace those characters at all.

eopo, 2023-08-30:
We used that, because multimon introduced a fully new format for flex messages that we couldn't parse. I think they integrated a new flag, that makes it use the old format, I'm not sure on that one though