pagermon/pagermon #554: Airspy

aleluja123, 2023-04-13:
Hey.
Any chance to use Airspy-mini instead of RTL-SDR ?

geelongmicrosoldering, 2023-04-13:
It might be possible, but its outside of the scope of pagermon. Prior to pagermons reader.js, messages come from multimon-ng, which is typically fed a pipe from rtl_fm.

Youd have to replace rtl_fm with airspy_rx i guess. Maybe pipe it through sox.

Your best bet would be to ask on multimon-ng GH

eopo, 2023-04-17:
Pagermon can theoretically process ANYTHING, you get to pipe into it. The only thing that's important if you want our reader.js-Script is, that the format of whatever you pipe to it is the one of multimon-ng. You could write your own tool that pushes messages to the pagermon-server API though, if you'd need something entirely different.

closed 2024-11-01 by eopo