pagermon/pagermon #557: Change columns visible in mobile view

AndrewJ1990, 2023-05-14:
**Is your feature request related to a problem? Please describe.**

When in mobile view using an iOS device (unable to check with android) the columns visible are date/time, icon, agency and message 

**Describe the solution you'd like**

To be able to choose which columns are visible, eg show the alias instead of agency. 

**Describe alternatives you've considered**

Columns are just the same as desktop view when device is rotated to landscape. 

**Additional context**

Requesting desktop view in any mobile browser doesn't appear to help

geelongmicrosoldering, 2023-05-15:
Technically this comes down to the theme. The columns that are hidden are assigned a class that hides them (i think its noMobile, dont quote me).

You can edit the theme and change which columns are hidden by moving that class to a different column. It may be possible to have some check boxes in the back end, but then every theme would have to be written with that in mind. Im not sure if thats the best approach given a theme could be created that doesnt use columns at all

AndrewJ1990, 2023-05-15:
> Technically this comes down to the theme. The columns that are hidden are assigned a class that hides them (i think its noMobile, dont quote me).
>
>
>
> You can edit the theme and change which columns [...]

Thanks, Do you know where in the code I could find that to edit? I've had a bit of a look but haven't found it so far

DanrwAU, 2023-09-27:
In index.ejs look for lines like this

```
th class="noMobile" style="text-align: center;">Capcode</th>
```

The part that makes them hidden on mobile is the `class="noMobile" ` if you remove this they will show on mobile

eopo, 2024-04-14:
#583 did hardcode one of that. We should probably make that configurable in options.