pagermon/pagermon #559: Port Number

killertez, 2023-06-16:
hello just wondering if you be able to add port change from 3000 to what ever port number we would like to use

geelongmicrosoldering, 2023-06-16:
This has been [asked before](https://github.com/pagermon/pagermon/issues/162). You actually can already. There's at least 2 ways to do this.

- You can edit process.json (you'll find the default source code in [process-default.json](https://github.com/pagermon/pagermon/blob/a5c983d7d1445cf10ea3e2aa3ad77dae11049476/server/process-default.json#L20)) and add a PORT variable to env, like this:

```
  "env": {
    "NODE_ENV": "production",
    "HOSTNAME": "",
    "PORT": 3050,
    "USE_COOKIE_HOST": false,
    "APP_NAME": "pagermon"
  }
```

- or you can modify the source code under [app.js](https://github.com/pagermon/pagermon/blob/a5c983d7d1445cf10ea3e2aa3ad77dae11049476/server/app.js#L94), replacing 3000 with your desired port

If your plan is to change it to port 80 or run it in production, you are probably better off just reading the Tutorial that was written, and use a reverse proxy [as mentioned here](https://github.com/pagermon/pagermon/wiki/Tutorial---Production-Ready-Ubuntu,-PM2,-Nginx-Reverse-Proxy,-Let's-Encrypt-SSL,-Pagermon-server#user-content-configuring-nginx-reverse-proxy-and-setting-up-lets-encrypt). Note: You don't have to use lets encrypt, and you don't have to have SSL.

A non-SSL version of the nginx server block might look something like this
```
server {
    listen 80 default_server;
    server_name pagermonhome.local www.pagermonhome.local; 
    location / {
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-NginX-Proxy true;
        proxy_set_header Upgrade $http_upgrade;
        proxy_set_header Connection "Upgrade";
        proxy_pass http://localhost:3000/;
        proxy_ssl_session_reuse off;
        proxy_set_header Host $http_host;
        proxy_cache_bypass $http_upgrade;
        proxy_redirect off;
 }
}
```

killertez, 2023-06-16:
oh ok awesome i didn't think anyone has asked that question before

closed 2023-06-30 by DanrwAU