pagermon/pagermon #560: pm2 status displays 2 readers help please

bertonumber1, 2023-06-21:
Hi , 

Ive messed something up and there is nothing appearing in the webserver .
I dont know how to recover from this issue . I notices there are 2 x readers running when running pm2 status.
Any ideas?
Please accept my apologies if i havent filled the issue form out correctly .
Thanks for your amazing work
Regards 

┌────┬────────────────────┬──────────┬──────┬───────────┬──────────┬──────────┐
│ id │ name               │ mode     │ ↺    │ status    │ cpu      │ memory   │
├────┼────────────────────┼──────────┼──────┼───────────┼──────────┼──────────┤
│ 1  │ pagermon           │ cluster  │ 0    │ online    │ 0%       │ 62.0mb   │
│ 0  │ reader             │ fork     │ 0    │ online    │ 0%       │ 2.7mb    │
│ 2  │ reader             │ fork     │ 920  │ online    │ 0%       │ 2.7mb    │
└────┴────────────────────┴──────────┴──────┴───────────┴──────────┴──────────┘

geelongmicrosoldering, 2023-06-21:
Your pm2 status shows that its running. Your second reader has apparently restarted 920 times, but that component is totally seperate to the server/web component.

What do you actually get in a browser?

bertonumber1, 2023-06-21:
Hi thanks for your rapid response .

I dont get anything at all in the browser.

Just now i ran 

"pm2 stop 2 "

then ran "pm2 delete 2"

So the table is back to normal at least ....

rtl_fm is sitting at 153.350M using an sdr-blog v3 dongle . i tried running the dongle in sdr# before ;osting and its working okay so i know ive goosed something i think during install when i ran 'pm2 startup' or 'pm2 start reader.sh'  twice by accident in the client folder.

so just now webserver is running , im logged into server and the rtl_fm and multimon are running , but sweet nothing by way of messages in the browser except :   

'No messages found!
Have you loaded any message data yet?

If you believe you reached this message by error, [log an issue.](https://github.com/pagermon/pagermon/issues)'

Regards

geelongmicrosoldering, 2023-06-21:
ok so what you are saying is the browser totally displays the page, and you aren't getting messages.

what you had explained sounded like you were getting a completely blank page with no menu, no text, and no source code, seeing as that would be "nothing at all", or some kind of error, which is why i asked.

not every area has the same paging network, frequency, protocol etc. what you need to do is stop your reader again
`pm2 stop reader`

and then run your command again without using pagermon at all, so, just pipe rtl_fm into multimon.
something like this:
`rtl_fm -d 0 -E dc -F 0 -A fast -f 153.3500M -s22050 - |
multimon-ng -q -b1 -c -a POCSAG512 -f alpha -t raw /dev/stdin`

if you aren't getting messages in your shell, something in your config for rtl_fm or multimon-ng is incorrect. but its an issue outside of pagermon.

If you are getting messages and when you pipe them into reader.js they dont end up in the server, did you follow [the wiki here](https://github.com/pagermon/pagermon/wiki/Client-Installation)?

bertonumber1, 2023-06-21:
**''ok so what you are saying is the browser totally displays the page, and you aren't getting messages.''**

_Yes,  apologies i meant in respect of messages there was nothing displaying_  

> If you are getting messages and when you pipe them into reader.js they dont end up in the server, did you follow [the wiki here](https://github.com/pagermon/pagermon/wiki/Client-Installation)?

I have ran it a few times now just by rtl_fm and multimon-ng and i get the pocsag messages . Everything was followed to the letter . I have even changed the secret phrase and restarted several times and always the ''no messages'' in the server.

I have checked over the reader.sh file and the only thing i can think causing the issue is ive added all the decoder modes to the multimon-ng section of the reader.sh
eg. -a POCSAG512 -a POCSAG1200 -a POCSAG2400 -a FLEX

Not sure if this could be the cause but i guess the only thing left is to start afresh .

Thanks for your time 

Regards

DanrwAU, 2023-06-30:
Support is via Discord - this is not an application issue but a configuration one.

closed 2023-06-30 by DanrwAU