pagermon/pagermon #561: Notification of alarms via MQTT

scripthase, 2023-06-29:
Hello everyone,

I would like to use Pagermon at home to trigger my smart home for alarms and calls.

A lot is done with MQTT in the smart home. So it would be practical if Pagermon could also forward alarms to my smart home server via MQTT.

So far I haven't found a solution here and I haven't really had an idea either.

Maybe someone can help me here and has an idea? Maybe that's easy to do with a triggered script or something?

I would be very grateful for any help.

Best regards

DanrwAU, 2023-06-30:
You could probably do it via the run external program plugin, otherwise best bet is to write your own plugin. There are quite a few node modules on NPM that could be used to build one.

ChoffaH, 2023-11-12:
@scripthase I just created a plugin for MQTT (also with home assistant discovery support) in my fork. I would love to get some feedback as I haven't work very much with MQTT before. I'm currently testing it on my machine and will create a PR when I feel like it's working as I want.
You can find it here: https://github.com/ChoffaH/pagermon/tree/develop

denperss11, 2024-03-03:
Could you share your build ? I would love to have this feature also! 

> [scripthase](/scripthase)

ChoffaH, 2024-03-04:
> Could you share your build ? I would love to have this feature also!
> 
> > [scripthase](/scripthase)

I don't really have a build to share. If you go to the link in my post you can download the code and build it yourself 🙂

s3m1s0n1c, 2024-03-04:
@denperss11 
Just wget these files from @ChoffaH and put them in ya pagermon/server/plugins folder
https://raw.githubusercontent.com/ChoffaH/pagermon/develop/server/plugins/MQTT.js
https://raw.githubusercontent.com/ChoffaH/pagermon/develop/server/plugins/MQTT.js

Thanks @ChoffaH Will try it out soon.. Wanna make a Pagermon stats in Node-Red so mqtt makes it easier then doing DB query's

PR #584 (2024-03-15): MQTT plugin
# Description

A plugin that sends messages to an configurable MQTT broker. It requires the "mqtt" node package, added to package.json.
It also includes optional support for Home Assistant auto discovery via MQTT.

Fixes #561 

## Type of change

- [x] New feature (non-breaking change which adds functionality)

# How Has This Been Tested?

I have tested this locally on my server for a couple of weeks and it's sending messages. Don't know how to test this more?

# Checklist:

- [x] My code follows the style guidelines of this project 
- [x] I have performed a self-review of my own c [... cut]
closed 2024-04-14 by eopo