pagermon/pagermon #562: HTTPS

killertez, 2023-06-30:
is there a way i can turn on HTTPS

RIKIKU, 2023-06-30:
I believe the most common way to do this is to use a Reverse Proxy like Nginx. 
Personally, I use [LinuxServer.io/SWAG](https://github.com/linuxserver/docker-swag) which also automates the SSL renewals for free via Let's encrypt. 
The benefit of using a reverse proxy is that you can put any of your web apps behind it and use SSL for any of them.

killertez, 2023-06-30:
this looks complicated... is there a way to auto install it with a script.

RIKIKU, 2023-06-30:
hmm, have you hosted an application that implements HTTPS before? There's usually some complexity somewhere and I thought SWAG was one of the simpler implementations.

killertez, 2023-06-30:
i haven't used that one before

RIKIKU, 2023-06-30:
Which one have you used?

killertez, 2023-06-30:
i normally have the programs do it for my auto

RIKIKU, 2023-06-30:
That's odd, are they just using self signed certificates? Can you tell me which apps specifically do this?

killertez, 2023-06-30:
i have a control panel for my websites that allow me to turn it on or off....
but i cant do that for my server

RIKIKU, 2023-06-30:
Are your websites like a square space or wix type deal?

killertez, 2023-06-30:
no i have a dedicated website host

RIKIKU, 2023-06-30:
A lot of the complexity of managing SSL certificates and configuring the sites is hidden away when you use paid services. 
The complexity is still there, it's just hidden and it's difficult to automate on a project like pagermon. 
It's the reason not many projects like this one try to implement SSL and just use a reverse proxy (RP), as this type of thing is bread and butter for an RP and there are many who have done a great job of it already. 

If you're not keen on having a crack at a self hosted RP, you might want to check out [Cloudflare Tunnel](https://www.cloudflare.com/products/tunnel/). As I understand it, it will publish your app on the Internet, and it will handle SSL on the public side for you. It should be substantially easier to configure.  To be clear, I have no experience using this product and I don't know if it's free, but a few friends of mine have tried it and reckon it's great.

killertez, 2023-06-30:
i have tried that wouldn't work

killertez, 2023-06-30:
i just keep getting This site can’t provide a secure connection

DanrwAU, 2023-06-30:
Support for pagermon is via Discord, this isn't appropriate for a github issue. Post over in discord and the community may be able to assist, however this is not pagermon specific support, it is server administration.

closed 2023-06-30 by DanrwAU