pagermon/pagermon #564: Text bug

112bzo, 2023-07-29:
**Describe the bug**
texts bug.

I see in the admin "Grip 3" but in the overview i see Gr**ee**p 3

[image] [image]

Its a text bug i think. 

Version : PagerMon 0.3.12-beta

geelongmicrosoldering, 2023-07-29:
There's no bug.

You are using Chrome. Chrome asked if you want to translate English to Dutch. You clicked yes.

English: Grip = Dutch: Greep

Turn off google translate. This isn't an issue with pagermon

112bzo, 2023-07-30:
Oh i'am stupid....... Tnx thats te isseu

closed 2023-07-30 by 112bzo