pagermon/pagermon #565: Raspberry 4 Installation Issues

leecollingsco, 2023-07-31:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Unable to screenshot, so I've copied and pasted my terminal output.
```
lee@raspberrypi:~/pagermon $ sudo npm install npm@latest -g
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'npm@9.8.1',
npm WARN EBADENGINE   required: { node: '^14.17.0 || ^16.13.0 || >=18.0.0' },
npm WARN EBADENGINE   current: { node: 'v12.22.12', npm: '7.5.2' }
npm WARN EBADENGINE }

added 1 package, and audited 250 packages in 9s

28 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
lee@raspberrypi:~/pagermon $ sudo npm install pm2 -g
ERROR: npm v9.8.1 is known not to run on Node.js v12.22.12.  This version of npm supports the following node versions: `^14.17.0 || ^16.13.0 || >=18.0.0`. You can find the latest version at https://nodejs.org/.

ERROR:
/usr/local/lib/node_modules/npm/lib/utils/exit-handler.js:19
  const hasLoadedNpm = npm?.config.loaded
                           ^

SyntaxError: Unexpected token '.'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at module.exports (/usr/local/lib/node_modules/npm/lib/cli-entry.js:15:23)
    at module.exports (/usr/local/lib/node_modules/npm/lib/es6/validate-engines.js:39:10)
    at module.exports (/usr/local/lib/node_modules/npm/lib/cli.js:4:31)
lee@raspberrypi:~/pagermon $ cd server
lee@raspberrypi:~/pagermon/server $ npm install
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'triple-beam@1.4.1',
npm WARN EBADENGINE   required: { node: '>= 14.0.0' },
npm WARN EBADENGINE   current: { node: 'v12.22.12', npm: '7.5.2' }
npm WARN EBADENGINE }
npm WARN deprecated @npmcli/move-file@1.1.2: This functionality has been moved to @npmcli/fs
npm WARN deprecated request-promise@4.2.2: request-promise has been deprecated because it extends the now deprecated request package, see https://github.com/request/request/issues/3142
npm WARN deprecated urix@0.1.0: Please see https://github.com/lydell/urix#deprecated
npm WARN deprecated source-map-url@0.4.1: See https://github.com/lydell/source-map-url#deprecated
npm WARN deprecated resolve-url@0.2.1: https://github.com/lydell/resolve-url#deprecated
npm WARN deprecated source-map-resolve@0.5.3: See https://github.com/lydell/source-map-resolve#deprecated
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated node-pre-gyp@0.11.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm WARN deprecated fsevents@2.1.3: "Please update to latest v2.3 or v2.2"
npm WARN deprecated request@2.88.0: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated @babel/polyfill@7.12.1: 🚨 This package has been deprecated in favor of separate inclusion of a polyfill and regenerator-runtime (when needed). See the @babel/polyfill docs (https://babeljs.io/docs/en/babel-polyfill) for more information.
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated node-uuid@1.4.8: Use uuid module instead
npm WARN deprecated snekfetch@3.6.4: use node-fetch instead
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated debug@3.2.6: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated slack@11.0.2: this library is no longer maintained; if you are interested in taking this package on let us know
npm WARN deprecated brute-knex@4.0.1: Package no longer supported. Contact Support at https://www.npmjs.com/support for more info.
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated babel-eslint@9.0.0: babel-eslint is now @babel/eslint-parser. This package will no longer receive updates.
npm WARN deprecated debug@4.1.1: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated axios@0.19.2: Critical security vulnerability fixed in v0.21.1. For more information, see https://github.com/axios/axios/pull/3410
npm WARN deprecated angular@1.8.3: For the actively supported Angular, see https://www.npmjs.com/package/@angular/core. AngularJS support has officially ended. For extended AngularJS support options, see https://goo.gle/angularjs-path-forward.
npm WARN deprecated oracledb@4.2.0: Update to node-oracledb 5
npm WARN deprecated discord.js@11.6.4: no longer supported
npm WARN deprecated @discordjs/uws@10.149.0: Package no longer supported. Contact Support at https://www.npmjs.com/support for more info.
npm WARN deprecated core-js@2.6.12: core-js@<3.23.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Some versions have web compatibility issues. Please, upgrade your dependencies to the actual version of core-js.

added 1206 packages, and audited 1207 packages in 8m

88 packages are looking for funding
  run `npm fund` for details

34 vulnerabilities (12 moderate, 13 high, 9 critical)

To address issues that do not require attention, run:
  npm audit fix

To address all issues possible (including breaking changes), run:
  npm audit fix --force

Some issues need review, and may require choosing
a different dependency.

Run `npm audit` for details.
lee@raspberrypi:~/pagermon/server $ export NODE_ENV=production
lee@raspberrypi:~/pagermon/server $ pm2 start process.json
bash: pm2: command not found
lee@raspberrypi:~/pagermon/server $ 
```

**Environment information:**
 - OS: PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

 - Other details:

**Additional context**
There's another issue someone's posted saying the installation steps should be made a little clearer for non-tech savvy users. And I agree with this. There seems to be an issue with the current installation steps, as from the first try, it seems they don't work and I'm unable to install this with nowhere to go?

leecollingsco, 2023-07-31:
The fatal issue seems to be with installing pm2, which for some reason doesn't work according to the installation steps.

```
lee@raspberrypi:~/pagermon/server $ sudo npm i -g pm2
ERROR: npm v9.8.1 is known not to run on Node.js v12.22.12.  This version of npm supports the following node versions: `^14.17.0 || ^16.13.0 || >=18.0.0`. You can find the latest version at https://nodejs.org/.

ERROR:
/usr/local/lib/node_modules/npm/lib/utils/exit-handler.js:19
  const hasLoadedNpm = npm?.config.loaded
                           ^

SyntaxError: Unexpected token '.'
    at wrapSafe (internal/modules/cjs/loader.js:915:16)
    at Module._compile (internal/modules/cjs/loader.js:963:27)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at module.exports (/usr/local/lib/node_modules/npm/lib/cli-entry.js:15:23)
    at module.exports (/usr/local/lib/node_modules/npm/lib/es6/validate-engines.js:39:10)
    at module.exports (/usr/local/lib/node_modules/npm/lib/cli.js:4:31)
lee@raspberrypi:~/pagermon/server $ 
```

So, does this mean, because the fault seems to be with the npm script, that nothing can be done? Or what exactly is this trying to tell us?

This is probably what other users mean by saying it's not helpful for non-tech savvy people. There's no clear instruction on what to do here, or how this can be resolved from what I can see?

eopo, 2023-08-30:
Oh, I've encountered a similar problem myself.
You are using an outdated version of node.js, which is not capable of running code some of the libraries we (and nom) use.
Additionally, you updated npm itself to the latest version, which also does not run on your version of node. 
Please update node to the latest LTS.

closed 2023-09-16 by marshyonline