pagermon/pagermon #571: High Memory Usage on Alias Admin page

DanrwAU, 2023-09-17:
**Describe the bug**
Large memory usage when loading the "Alias" admin page when a large number of aliases are defined. 

Tied to fontawesome. does not happen when fontawesome is disabled. suspect number of icons loading has a significant impact

**To Reproduce**
Load a significant number of aliases in with Icons defined. 
Load Alias page, browser will consume high amounts of memory, with a large numebr of aliases this will also crash the browser with an "out of memory" error

**Expected behavior**
Works

**Additional context**
Possible solutions https://fontawesome.com/docs/web/add-icons/svg-symbols

DanrwAU, 2023-09-27:
fixed in #572 - rolled back to FontAwesome 4 - breaking change

aaronchitaranjan, 2023-12-28:
@DanrwAU possible solution that seems to work for me, is to change the 'Technology' of the FA Kit to `Web Fonts` instead of `SVG`. I'm getting dramatically improved alias page loads with 700+ aliases

bullseye555, 2023-12-28:
@DanrwAU and @aaronchitaranjan - I've tried this too (after seeing the suggestion!) with 9000 odd capcodes. 

Only issue has been some icons not loading, but for the sake of having the _Alias_ page loading, I'll happily change the one icon that's not loading!

aaronchitaranjan, 2023-12-28:
> @DanrwAU and @aaronchitaranjan - I've tried this too (after seeing the suggestion!) with 9000 odd capcodes.
> 
> Only issue has been some icons not loading, but for the sake of having the _Alias_ page loading, I'll happily change the one icon that's not loading!

Good to hear. Might be worth adding it to the README/settings page if more people report similar positive results 🤷

Shaggs, 2024-01-09:
Have the same issue. Opening alias crashes with 1070 CAPcodes 😳.

Refreshing CAPcodes is also an issue. 

Have issues on rpi and i5 16gb RAM (testing to see if it was a memory issue).

Looks to be a limitation with node?

aaronchitaranjan, 2024-01-09:
> Have the same issue. Opening alias crashes with 1070 CAPcodes 😳.
> 
> Refreshing CAPcodes is also an issue.
> 
> Have issues on rpi and i5 16gb RAM (testing to see if it was a memory issue).
> 
> Looks to be a limitation with node?

Did changing the icon pack technology help?

Shaggs, 2024-01-09:
I had it set to `SVG` now changed to `web font`. 

Have issues with no icons set

aaronchitaranjan, 2024-01-09:
> I had it set to `SVG` now changed to `web font`.

Let us know how you go :)

Shaggs, 2024-01-09:
> > I had it set to `SVG` now changed to `web font`.
> 
> Let us know how you go :)

Yeah that fixed it. 👍

s3m1s0n1c, 2024-01-09:
I fixed it by using
<link rel="stylesheet" href="https://cdnjs.cloudflare.com/ajax/libs/font-awesome/4.7.0/css/font-awesome.css" integrity="sha512-5A8nwdMOWrSz20fDsjczgUidUBR8liPYU+WymTZP1lmY9G6Oc7HlZv156XqnsgNUzTyMefFTcsFH/tnJE/+xBg==" crossorigin="anonymous" referrerpolicy="no-referrer" /> in header.ejs
instead of the one in there..

gg33fr, 2024-02-27:
Hello, when I add aliases, and I do a refresh I have the circle that turns on the page and the refresh does not take place.
This actually looks like a memory problem.
I added the command line given by s3m1s0n1c, maybe I did it wrong.
Can you put the entire header.ejs file
I am on the PagerMon 0.3.11-beta version with a pi 4 - 1GB
Thanks

Shaggs, 2024-03-10:
> Hello, when I add aliases, and I do a refresh I have the circle that turns on the page and the refresh does not take place. This actually looks like a memory problem. I added the command line given [...]

Just change to Web font..  much easier

PR #574 (2023-09-29): Fa4 rollback
# Description

Rollback to FontAwesome 4 to fix performance issues

Fixes #571 

## Type of change

Please delete options that are not relevant.

- [x] Bug fix (non-breaking change which fixes an issue)
- [x] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [x] This change requires a documentation update

# How Has This Been Tested?

Tested on multiple servers

# Checklist:

- [x] My code follows the style guidelines of this project
- [x] I have performed a self-review of my own code
- [x] I have commented my code, particularly in  [... cut]