pagermon/pagermon #575: How to simply upgrade from old version  0.3.5-beta

techofix4u, 2023-10-17:
Hi All,
A while ago (Pagermon version 0.3.5-beta), I setup Pagermon. It has been running great. It took a lot of effort to get it going, but I was pleased to get there and have been successful decoding since.
I'm not too good with Linux and I want to upgrade my current install from 0.3.5-beta to the latest version.
Is there a simple way to do this without losing my reader.sh config, my long file splitting config and all of my cap codes and descriptions?
If someone could supply the terminal commands that would be perfect.
Please let me know if you need further information on my install.
Many thanks.

f5gkw, 2024-05-30:
Idem, same as you :)

erikklavora, 2024-09-30:
> Hi All, A while ago (Pagermon version 0.3.5-beta), I setup Pagermon. It has been running great. It took a lot of effort to get it going, but I was pleased to get there and have been successful [...]

Hi, do you have a how to of what you did to get Pagermon working? I'm scratching my head but not working... i mean, it works from client toserver but not when i try to pass pocsag messages from rtl_fm througm multimon-ng to node reader.js

DanrwAU, 2024-10-14:
Support is available in #support on the project discord.

closed 2024-10-14 by DanrwAU