pagermon/pagermon #588: All New Messages are displayed on Filtered results

bullseye555, 2023-12-26:
**Describe the bug**
When a filter is applied and on Page 1 of the filter, _all_ new messages are displayed in the filter list
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the main page
2. Apply a filter (in the screenshot above, this is for _FS90 Patterson River_)
3. Stay on the page and wait for new decoded messages to show
4. Once new messages are decoded, they will appear

**Expected behavior**
Either
1. Only new messages for the filtered results should show (so, in the above scenario, only new messages for FS90); or,
2. No new messages should appear

**Environment information:**
 - OS: Pagemon Pi Image - with manual upgrade to v0.3.13
 - Other details: Using custom themes, but have confirmed this also occurs on the _Default_ theme
[image]

bullseye555, 2024-01-05:
Updating this - I've also noticed that that when this is filtered, the _new_ decodes don't lengthen the result page, but replace the filtered decodes

EG: this filter has 7 results
[image]

Instead of the next 13 results appearing on top and the original 7 moving down the page (with the default 20 results shown), it will replace them with the next 7 decodes (with the process continuing for every new decode)
[image]

Additional debug - this appears to only occur when filtering by _Alias_
Filtering by Agency, Capcode, or specific custom text does not appear to be an issue (at least, in my testing) - and I don't have multiple sources to see if filtering by Source is an issue

eopo, 2024-04-19:
> Updating this - I've also noticed that that when this is filtered, the new decodes don't lengthen the result page, but replace the filtered decodes
Is gonna be fixed by #597

I'll have a look for the other issue. Currently, we do send all messages via socket and then filter client side, if we want them to be shown. That seems to not work in your case.

DanrwAU, 2024-10-14:
Fixed by #597

closed 2024-10-14 by DanrwAU