pagermon/pagermon #589: Error 502

theaustinayers, 2024-02-04:
Hello,

On a clean install of Ubuntu 22.04.3 I am now receiving an Error 502 upon enabling PDW Mode, Show Admin, and Hide Capcodes.

Upon reboot, NGINX is hanging on Error 502. I manually killed the PM2 instance and restarted it

WizRd0, 2024-02-18:
To get your install back working, assuming it's directly related to the config changes you made you can edit pagermon/server/config/config.json and change the settings back in the messages config section then restart pagermon with "pm2 restart pagermon".

```
"pdwMode": false,
"adminShow": false,
"hideCapcode": false,
```

If that doesn't work some logs from "pm2 logs pagermon" will likely help the bug report.

eopo, 2024-04-19:
Are you able to recreate that situation and provide logs?

closed 2024-10-14 by DanrwAU