pagermon/pagermon #590: Pagermon problem

MeteoGenderen, 2024-02-16:
I have been trying to get Multimon and Pagermon working for a while now, but I keep running into the same problems.
Receiving 112 messages works and Pagermon also seems to work, but there is no way to view the server in HTML. (http://192.168.178.171:3000)
Who can help me further?
[image]

WizRd0, 2024-02-18:
As pm2 status shows that pagermon is online but has an uptime of 0s, often if there are errors with the server it won't show for a few seconds.  Is it still "online".  If it's been running for a minute and it's still online use the command "sudo ss -ntalp" to check if port 3000 is listening.

If it's not running after a few seconds use "pm2 logs" to have a look why it's dying, often it's something simple like sqlite3 isn't installed but your config in pagermon/server/config/config.json is set to sqlite3.

Assuming it's running and it's listening I would next be looking at a firewall blocking it, AFAIK (which is information from 2 years ago) raspbian doesn't have a firewall enabled by default, but if you have enabled one its likely UFW, in which case you should see blocks in /var/log/syslog - use "sudo grep -i UFW /var/log/syslog".  "sudo ufw allow 3000" should get you going.

Other than these few things if there is another firewall in your network that's doing the blocking you will need to investigate that.

WizRd0, 2024-03-13:
As you now have a working installation please do not forget to close the issue.

closed 2024-04-14 by eopo